winterheart/broadcom-bt-firmware #1: Can't use string ("") as a HASH ref while "strict refs" in use at tools/bt-fw-converter.pl line 123, <$in> line 1

moonlies, 2017-01-10:
I tested this on a clients device (Ubuntu 16.04 BC) using the package [WIDCOMM® Bluetooth Windows® Software for Windows 10 & 8, 64-Bit](https://docs.broadcom.com/docs/12358522). The Bluetooth Device is BC43142 a0 acer combo DeviceNR: 0489:e055
After populating the source/ directory with all .inf and .hex files from the archive and running generate_files.sh i get the error:

"Can't use string ("") as a HASH ref while "strict refs" in use at tools/bt-fw-converter.pl line 123, <$in> line 1"

Also no files are generated in bcrm/ directory.

winterheart, 2017-01-10:
Hello.
You don't have to run this tool by self, I already done that. If you only need hcd file, just copy required file from brcm folder (BCM43142A0-0489-e055.hcd).
But yes, currently converting from raw hex-files is little quirky.

winterheart, 2017-02-17:
This issue should be now fixed.

closed 2017-02-17 by winterheart