winterheart/broadcom-bt-firmware #3: Missing firmware BCM4354A2-13d3-3485.hcd for BCM4354/BCM4356 chip

rauferd, 2017-07-26:
My bluetooth fails with
`bluetooth hci0: Direct firmware load for brcm/BCM4354A2-13d3-3485.hcd failed with error -2`
The missing firmware file is not available here, so I assume it is correct to open a new issue for this.

The chip is installed in a Wortmann / terra mobile 360-11 netbook. Windows drivers can be downloaded from this page:
https://www.wortmann.de/de-de/systeminformation/R5105360.aspx
(in German; look for the item "Terra Mobile 360-11 Bluetooth Treiber")
However, there seems to be no .hex file, only some .bin files. Any hint on how to convert them to .hcd for use with linux (and for inclusion in this repository)?

rauferd, 2017-07-26:
Just out of curiosity, i tried renaming the firmware file BCM435**6**A2-13d3-3485.hcd to BCM435**4**A2-13d3-3485.hcd and rebooted.
Now the firmware loads (no error messages) and I can still scan for new devices, but I can no longer complete the pairing process or connect to any device, whether previously paired or not.

winterheart, 2017-07-26:
Hello. 

First of all, firmware from this repository will not work with your device, it's total different chipset family.
But, as I can see, *.bin files are already in hcd format. Question is which of them can be for your device.
Can you please dump output from `lsusb` command?

rauferd, 2017-07-26:
Thank you for the quick reply. Please find below the output of the lsusb and lspci commands. I guess the actual bluetooth device is part of the Broadcom wifi chip (last line).
There is a file called `4356a2rtecdc.bin`, but my kernel actually asks for `BCM4354A2-13d3-3485.hcd`, see above. There is no file named `4354*.bin`.

> $ uname -a  
> Linux Terra-N360-11 4.11.0-1-amd64 #1 SMP Debian 4.11.6-1 (2017-06-19) x86_64 GNU/Linux
> $ lsusb
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 004: ID 13d3:3485 IMC Networks 
> Bus 001 Device 003: ID 04f2:b54f Chicony Electronics Co., Ltd 
> Bus 001 Device 006: ID 045e:0084 Microsoft Corp. Basic Optical Mouse
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> $ lspci
> 00:00.0 Host bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register (rev 35)
> 00:02.0 VGA compatible controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller (rev 35)
> 00:0b.0 Signal processing controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Power Management Controller (rev 35)
> 00:13.0 SATA controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SATA Controller (rev 35)
> 00:14.0 USB controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series USB xHCI Controller (rev 35)
> 00:1a.0 Encryption controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Trusted Execution Engine (rev 35)
> 00:1b.0 Audio device: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series High Definition Audio Controller (rev 35)
> 00:1c.0 PCI bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCI Express Port #1 (rev 35)
> 00:1c.1 PCI bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCI Express Port #2 (rev 35)
> 00:1f.0 ISA bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCU (rev 35)
> 00:1f.3 SMBus: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx SMBus Controller (rev 35)
> 01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)
> 02:00.0 Network controller: Broadcom Limited BCM4356 802.11ac Wireless Network Adapter (rev 02)

winterheart, 2017-07-27:
That's weird. You have "ID 13d3:3485 IMC Networks" that threats as BCM4354A2, but it should be BCM4356A2. lspci shows too that there should be BCM4356. Can you please post output of these commands:

- `dmesg`
- `lspci -nnvv | grep -A12 Broadcom`
- `hciconfig -a`

rauferd, 2017-07-27:
This is the full output of dmesg: [https://pastebin.com/SES6sMyS](https://pastebin.com/SES6sMyS)
Note: After booting, I tried to connect one wireless bluetooth presenter to the computer using blueman.

> $ lspci -nnvv | grep -A12 Broadcom
> 02:00.0 Network controller [0280]: Broadcom Limited BCM4356 802.11ac Wireless Network Adapter [14e4:43ec] (rev 02)
> 	Subsystem: AzureWave BCM4356 802.11ac Wireless Network Adapter [1a3b:2217]
> 	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 314
> 	Region 0: Memory at 91400000 (64-bit, non-prefetchable) [size=32K]
> 	Region 2: Memory at 91000000 (64-bit, non-prefetchable) [size=4M]
> 	Capabilities: <access denied>
> 	Kernel driver in use: brcmfmac
> 	Kernel modules: brcmfmac

As far as I can tell, this is only the wifi (802.11) device, not the bluetooth device.

> $ hciconfig -a
> hci0:	Type: Primary  Bus: USB
> 	BD Address: F0:03:8C:9E:82:7A  ACL MTU: 1021:8  SCO MTU: 64:1
> 	UP RUNNING PSCAN 
> 	RX bytes:3310 acl:31 sco:0 events:126 errors:0
> 	TX bytes:6228 acl:31 sco:0 commands:99 errors:0
> 	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
> 	Link policy: RSWITCH SNIFF 
> 	Link mode: SLAVE ACCEPT 
> 	Name: 'N360-11'
> 	Class: 0x1c010c
> 	Service Classes: Rendering, Capturing, Object Transfer
> 	Device Class: Computer, Laptop
> 	HCI Version: 4.1 (0x7)  Revision: 0x2000
> 	LMP Version: 4.1 (0x7)  Subversion: 0x230f
> 	Manufacturer: Broadcom Corporation (15)

winterheart, 2017-07-27:
Well, here what I found. BCM4354A2 is actually BCM4356A2. It seems in Linux Kernel typo or bug.
So, let's try again to rename BCM4356A2-13d3-3485.hcd to BCM4354A2-13d3-3485.hcd, reboot and again `dmesg` and `hciconfig -a`

rauferd, 2017-07-27:
Ok. Did that. No more error messages during startup. But connections still did not work.

Now comes the good part: After consulting the linux wireless pages at https://wireless.wiki.kernel.org/en/users/drivers/brcm80211 (which are acutally about the corresponding wifi 802.11 device), I learned that in addition to the fireware some EFI NVRAM information may be required for this device. Following the instructions from that source, I executed the following commands as root:
```
mount -t efivarfs none /sys/firmware/efi/efivars
cat /sys/firmware/efi/efivars/nvram-74b00bd9-805a-4d61-b51f-43268123d113 > /lib/firmware/brcm/brcmfmac4356-pcie.txt
```
Notice the .txt extenstion vs. the .bin extension of the wifi firmware file.

Now after another reboot _everything works as expected_ (currently tested with the presentation remoate and different bluetooth speakers: HSP and A2DP).
Awesome!

Full dmesg output: https://pastebin.com/zPauTmbM

> $ hciconfig -a
> hci0:	Type: Primary  Bus: USB
> 	BD Address: F0:03:8C:9E:82:7A  ACL MTU: 1021:8  SCO MTU: 64:1
> 	UP RUNNING PSCAN 
> 	RX bytes:3122 acl:0 sco:0 events:361 errors:0
> 	TX bytes:37115 acl:0 sco:0 commands:360 errors:0
> 	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
> 	Link policy: RSWITCH SNIFF 
> 	Link mode: SLAVE ACCEPT 
> 	Name: 'N360-11-Fronczek'
> 	Class: 0x1c010c
> 	Service Classes: Rendering, Capturing, Object Transfer
> 	Device Class: Computer, Laptop
> 	HCI Version: 4.1 (0x7)  Revision: 0x20c5
> 	LMP Version: 4.1 (0x7)  Subversion: 0x230f
> 	Manufacturer: Broadcom Corporation (15)

closed 2017-07-27 by rauferd

winterheart, 2017-07-28:
Great to hear that. Thank you for useful information.