winterheart/broadcom-bt-firmware #4: Clarification needed: Licence and binary distribution

janhenke, 2017-09-17:
Hi,

I am interested in making this repository available to Gentoo Linux users as an ebuild, so they can use their Bluetooth hardware. One major issue is the licence. To me it is currently not clear whether this repository is correctly licensed by Broadcom to use the firmware files in binary form and if the licence also covers binary distribution to third parties.

I believe it is in common interest to get a clear understanding of this issue.

winterheart, 2017-09-18:
Hello.

There great domains of Unknowns. Broadcom don't ships license with drivers on Windows Update. I obtained current license from their official site when drivers has been available on it. According to it (I think) you cannot ships my tarball as source for ebuilds. I don't have any contacts with Broadcom to clear this issue.

But, you can use cab file from Windows Update (http://download.windowsupdate.com/c/msdownload/update/driver/drvs/2017/04/852bb503-de7b-4810-a7dd-cbab62742f09_7cf83a4c194116648d17707ae37d564f9c70bec2.cab for current version) and my script to convert hex to hcd directly on client machine.

manfred-h, 2020-11-25:
Has there been any update to this? I have been asked to supply a package for openSUSE, but looking at this git project/repository I doubt it's possible to generate anything which can be included with/built on OBS e.g. in the "hardware" project (details at [OBS](https://build.opensuse.org/project/show/hardware)). The license thing really is blocking it to incorporate your great package into a CICD pipeline on OBS, afaict, but ianal :)

winterheart, 2020-11-25:
Sorry, can't help here. I don't know any contacts on Broadcom side and can't negotiate on that matter. Doubt that can be resolved on my level.

manfred-h, 2020-11-26:
Understood! Since we aren't allowed to distribute the files in an RPM, I thought it might be possible to create an RPM for openSUSE which downloads the sources from your latest release, unpacks them and installs them to /lib/firmware via a shell script. So, each user does it himself when installing the RPM, which means we're not distributing the files :) Would this procedure be OK for you?

winterheart, 2021-05-13:
Oh, sorry for delay.

Yes, this approach can be applied.