winterheart/broadcom-bt-firmware #11: BCM20702A1-04ca-200c.hcd not working for [14e4:43b1]+[04ca:200c] mini-PCIe combo adapter

meden, 2019-04-13:
Hello,

I own a BCM4352+BCM20702A0 mini-PCIe WiFi+Bluetooth adapter (ids: `[14e4:43b1]+[04ca:200c]`). Following the naming convention, I downloaded and installed the `BCM20702A1-04ca-200c.hcd` from your repository.

The system detects the adapter and tries to load the firmware, but the loading fails (log excerpt):
```
Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
Bluetooth: hci0: command 0x1001 tx timeout
Bluetooth: hci0: BCM: Reading local version info failed (-110)
Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
```

Downloading the [Windows driver 12.0.0.9980](https://www.driverscape.com/files/2015/Bluetooth/broadcom_bluetooth_12_0_0_9980.zip) from [this page](https://www.hwdrivers.com/hw_ids/c71649), I came to extract the firmware file `BCM20702A1_001.002.014.1443.1494.hex`. Once converted to `hcd` format, renamed and installed, the Bluetooth module became functional (I could at least connect to my phone and transfer a file).

Hope this helps. Thanks for your work!

winterheart, 2019-04-14:
Hello.
Please attach info about your system:

```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb
```

meden, 2019-04-14:
Hello @winterheart, here they are (personal data redacted):
```Linux hostname 4.19.0-4-amd64 #1 SMP Debian 4.19.28-2 (2019-03-15) x86_64 GNU/Linux```
```
Bluetooth: Core ver 2.22
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO socket layer initialized
Bluetooth: hci0: BCM: chip id 63
Bluetooth: hci0: BCM: features 0x07
Bluetooth: hci0: BCM20702A
Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
bluetooth hci0: firmware: direct-loading firmware brcm/BCM20702A1-04ca-200c.hcd
Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Bluetooth: BNEP filters: protocol multicast
Bluetooth: BNEP socket layer initialized
Bluetooth: hci0: BCM20702A1 (001.002.014) build 1494
Bluetooth: hci0: Broadcom Bluetooth Device
Bluetooth: RFCOMM TTY layer initialized
Bluetooth: RFCOMM socket layer initialized
Bluetooth: RFCOMM ver 1.11
Modules linked in: fuse rfcomm cmac bnep bbswitch(OE) ipmi_devintf ipmi_msghandler snd_hda_codec_hdmi snd_hda_codec_realtek btusb btrtl btbcm snd_hda_codec_generic btintel bluetooth jitterentropy_rng drbg ansi_cprng rtsx_usb_ms memstick ecdh_generic intel_rapl asus_nb_wmi x86_pkg_temp_thermal asus_wmi intel_powerclamp sparse_keymap mxm_wmi joydev coretemp evdev kvm_intel i915 snd_hda_intel wl(POE) kvm uvcvideo snd_hda_codec irqbypass videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 crct10dif_pclmul crc32_pclmul snd_hda_core drm_kms_helper ghash_clmulni_intel snd_hwdep videobuf2_common cfg80211 intel_cstate snd_pcm videodev drm snd_timer intel_uncore serio_raw snd media intel_rapl_perf pcspkr soundcore sg iTCO_wdt rfkill i2c_algo_bit iTCO_vendor_support video wmi ac battery button pcc_cpufreq
```
```
03:00.0 Network controller [0280]: Broadcom Limited BCM4352 802.11ac Wireless Network Adapter [14e4:43b1] (rev 03)
	Subsystem: Broadcom Limited BCM4352 802.11ac Wireless Network Adapter [14e4:062e]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at dda00000 (64-bit, non-prefetchable) [size=32K]
	Region 2: Memory at dd800000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: <access denied>
	Kernel driver in use: wl
	Kernel modules: bcma, wl

04:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 06)
	Subsystem: ASUSTeK Computer Inc. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [1043:1277]
```
```
hci0:	Type: Primary  Bus: USB
	BD Address: XX:XX:XX:XX:XX:XX  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:3890 acl:0 sco:0 events:273 errors:0
	TX bytes:40641 acl:0 sco:0 commands:252 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
```
```
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 0bda:0139 Realtek Semiconductor Corp. RTS5139 Card Reader Controller
Bus 001 Device 004: ID 13d3:5710 IMC Networks UVC VGA Webcam
Bus 001 Device 003: ID 04ca:200c Lite-On Technology Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I'm noticing that the kernel detects device information matching working firmware filename I'm using:
* Kernel: `BCM20702A1 (001.002.014) build 1494`
* Working firmware: `BCM20702A1_001.002.014.1443.1494.hex`

The numbers match beside the `1443`. So I'm wondering:
# Why the BT modules developers changed the naming schema?
# There could be a chance that to the same USB VID&PID corresponds to multiple adapter revisions?

meden, 2019-04-14:
Wait, I just re-downloaded the firmware `BCM20702A1-04ca-200c.hcd` from your repo and checked the SHA sums: yours and the one of my manually generated firmware match:
```
c89d78041fcbd6c93099cfa86adf131f7234b624  BCM20702A1_001.002.014.1443.1494.hcd
d0b92bf5b88a5b427a5e7544d581682d70194fd8  BCM20702A1_001.002.014.1443.1494.hex
c89d78041fcbd6c93099cfa86adf131f7234b624  BCM20702A1-04ca-200c.hcd
```
So, not sure anymore about what happened...

winterheart, 2019-04-24:
Seems you have combined Wifi+Bluetooth device. Have you tried https://github.com/winterheart/broadcom-bt-firmware#notes-about-combined-wifibluetooth-devices ?

meden, 2019-04-25:
@winterheart thanks for the advice, although I'm not using EFI (it is a new installation on old laptop with dual boot, so I did not changed the partition table to GPT).

Both the wifi and bluetooth are working, now. The Bluetooth started working after extracting the firmware from the Windows driver, but then I compared the SHA1 (to add more information to this report) and found out that they were matching.
Maybe the fact it started working was just a coincidence (not sure anymore if I actually had a full power cycle on first attempt after installing the modules and the firmware for both devices).

Now, the only problem left is that using the fn key to cycle through WiFi+BT activation status is kinda broken (in Gnome 3.30, at least), but that's another story, likely related to the `asus_wmi` module and/or my laptop model (Asus K53SV), which uses the embedded controlled to directly manage these things, and its interaction with the gnome-settings-daemon.

I guess we can close this, if you don't see anything else off setting some alarm to you.

Thanks, and sorry for the noise.

winterheart, 2019-04-25:
OK, thanks.

closed 2019-04-25 by winterheart

kevchentw, 2020-05-15:
> Wait, I just re-downloaded the firmware `BCM20702A1-04ca-200c.hcd` from your repo and checked the SHA sums: yours and the one of my manually generated firmware match:
>
> ``` [...]

I encounter the same issue. After I extracting the firmware from .exe, bluetooth works, but the sha1 is same.