winterheart/broadcom-bt-firmware #15: Insignia USB Bluetooth Adapater BCM20702B

TragicWarrior, 2019-11-05:
From dmesg:

[336623.825141] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[336623.825458] bluetooth hci0: Direct firmware load for brcm/BCM20702B0-19ff-0239.hcd failed with error -2

Any support for this?  Purchased at BestBuy.

winterheart, 2019-11-05:
Yes, 

19ff:0239 | BCM20702B0-19ff-0239.hcd | Bestbuy 20702B0 micro
https://github.com/winterheart/broadcom-bt-firmware/blob/master/brcm/BCM20702B0-19ff-0239.hcd
You need place file into /lib/firmware/brcm and reboot system.

TragicWarrior, 2019-11-05:
Works great!  Thank you!

winterheart, 2019-11-08:
OK, glad to hear that.

closed 2019-11-08 by winterheart

parkerlreed, 2020-09-23:
This seems to have stopped working in 5.8.10

```
Bus 001 Device 007: ID 19ff:0239 Dynex Bluetooth 4.0 Adapter [Broadcom, 1.12, BCM20702A0]
[  799.789387] usb 1-10: new full-speed USB device number 9 using xhci_hcd
[  799.932032] usb 1-10: New USB device found, idVendor=19ff, idProduct=0239, bcdDevice= 1.12
[  799.932039] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  799.932043] usb 1-10: Product: BCM20702A0
[  799.932046] usb 1-10: Manufacturer: Broadcom Corp
[  799.932049] usb 1-10: SerialNumber: BC14EF9CDCC9
[  800.007143] kauditd_printk_skb: 21 callbacks suppressed
[  800.007145] audit: type=1130 audit(1600833263.884:256): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  800.048261] Bluetooth: hci0: BCM: chip id 73
[  800.049265] Bluetooth: hci0: BCM: features 0x07
[  800.065165] Bluetooth: hci0: BCM20702B
[  800.065168] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[  800.066175] Bluetooth: hci0: BCM20702B0 'brcm/BCM20702B0-19ff-0239.hcd' Patch
[  802.256002] Bluetooth: hci0: command 0x0a0a tx timeout
$ sudo hciconfig hci0 reset
Can't init device hci0: Connection timed out (110)
```

winterheart, 2020-09-23:
Please create another issue.