winterheart/broadcom-bt-firmware #16: brcm/BCM20702B0-0a5c-22be.hcd

soundarrajansurya, 2019-11-14:
root@kali:~# lsusb 
Bus 003 Device 003: ID 0a5c:22be Broadcom Corp. BCM2070 Bluetooth 3.0 + HS
root@kali:~#  dmesg | grep Blue
[  532.179295] Bluetooth: hci0: BCM: chip id 73
[  532.180254] Bluetooth: hci0: BCM: features 0x07
[  532.196252] Bluetooth: hci0: BCM20702B
[  532.197192] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[  532.197248] Bluetooth: hci0: BCM: Patch brcm/BCM20702B0-0a5c-22be.hcd not found

BCM20702B0-0a5c-22be.hcd file is not in the git.

winterheart, 2019-11-14:
Hello.

A bit strange device. Can you please share info on which hardware (model of notebook) you running on? And please provide output from commands:

```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb
```

soundarrajansurya, 2019-11-19:
hi @winterheart 
I'm using Kali linux in lenovo B490 laptop.

Output for the command you asked:-

root@kali:~# uname -a
Linux kali 5.2.0-kali2-amd64 #1 SMP Debian 5.2.9-2kali1 (2019-08-22) x86_64 GNU/Linux
root@kali:~# dmesg | grep -i Bluetooth
[   14.409385] Bluetooth: Core ver 2.22
[   14.409407] Bluetooth: HCI device and connection manager initialized
[   14.409411] Bluetooth: HCI socket layer initialized
[   14.409414] Bluetooth: L2CAP socket layer initialized
[   14.409418] Bluetooth: SCO socket layer initialized
[   14.719206] Bluetooth: hci0: BCM: chip id 73
[   14.720128] Bluetooth: hci0: BCM: features 0x07
[   14.736189] Bluetooth: hci0: BCM20702B
[   14.737202] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[   14.821265] bluetooth hci0: firmware: failed to load brcm/BCM20702B0-0a5c-22be.hcd (-2)
[   14.821506] bluetooth hci0: Direct firmware load for brcm/BCM20702B0-0a5c-22be.hcd failed with error -2
[   14.821514] Bluetooth: hci0: BCM: Patch brcm/BCM20702B0-0a5c-22be.hcd not found
[   92.748004] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   92.748008] Bluetooth: BNEP filters: protocol multicast
[   92.748019] Bluetooth: BNEP socket layer initialized
[   92.783234] Bluetooth: RFCOMM TTY layer initialized
[   92.783242] Bluetooth: RFCOMM socket layer initialized
[   92.783249] Bluetooth: RFCOMM ver 1.11
root@kali:~# lspci -nnvv | grep -A12 Broadcom
pcilib: sysfs_read_vpd: read failed: Input/output error
root@kali:~# hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 50:63:2A:02:77:3E  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN INQUIRY 
	RX bytes:9789 acl:61 sco:0 events:580 errors:0
	TX bytes:484742 acl:543 sco:1 commands:243 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'kali'
	Class: 0x1c010c
	Service Classes: Rendering, Capturing, Object Transfer
	Device Class: Computer, Laptop
	HCI Version: 4.0 (0x6)  Revision: 0x2000
	LMP Version: 4.0 (0x6)  Subversion: 0x410e
	Manufacturer: Broadcom Corporation (15)

root@kali:~# lsusb
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0a5c:22be Broadcom Corp. BCM2070 Bluetooth 3.0 + HS
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@kali:~# 

Help me to resolve the issue.

winterheart, 2019-11-19:
Try these fw: [bcm.zip](https://github.com/winterheart/broadcom-bt-firmware/files/3866106/bcm.zip)

Rename each of these to `/lib/firmware/brcm/BCM20702B0-0a5c-22be.hcd`, reboot and try to configure Bluetooth.

soundarrajansurya, 2019-11-26:
How can i place two files in same directory. And i tried with both the files but not working..

root@kali:~# dmesg | grep -i Bluetooth
[   14.275203] Bluetooth: Core ver 2.22
[   14.275228] Bluetooth: HCI device and connection manager initialized
[   14.275765] Bluetooth: HCI socket layer initialized
[   14.275767] Bluetooth: L2CAP socket layer initialized
[   14.275772] Bluetooth: SCO socket layer initialized
[   14.515494] Bluetooth: hci0: BCM: chip id 73
[   14.516502] Bluetooth: hci0: BCM: features 0x07
[   14.532575] Bluetooth: hci0: kali
[   14.533563] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[   14.654547] bluetooth hci0: firmware: direct-loading firmware brcm/BCM20702B0-0a5c-22be.hcd
[   17.240981] Bluetooth: hci0: command 0x0c03 tx timeout
[   25.240956] Bluetooth: hci0: BCM: Reset failed (-110)

winterheart, 2019-12-04:
Well, I don't see any compatible firmware. Officially Broadcom dropped support around 2011 and marked it as "no firmware needed".

closed 2019-12-04 by winterheart

xiaosuo, 2026-08-18:
You can follow the steps：
1. download cap from windowsupdate http://download.windowsupdate.com/d/msdownload/update/driver/drvs/2015/10/200006644_5f769d5f45e68840195d3bb24ef9d1ae2c791272.cab
2. install capextract（ https://www.cabextract.org.uk/cabextract-1.11.tar.gz ），use capextract to decompress the above file
3. grep -l 22BE | grep BCM20702B0 to get the correct hex file
4. install https://github.com/jessesung/hex2hcd.git ，and use it to convert the above hex file to BCM20702B0-0a5c-22be.hcd
5. place the hcd into /lib/firmware/brcm
6. modprobe -r btusb && modprobe btusb
7. check if the firmware is loaded successfully.

xiaosuo, 2026-08-25:
here is the firmware.

[BCM20702B0-0a5c-22be.hcd.gz](https://github.com/user-attachments/files/31425098/BCM20702B0-0a5c-22be.hcd.gz)