winterheart/broadcom-bt-firmware #17: BCM20702A1-0b05-17bc.hcd failed with error -2

vpomax, 2019-12-11:
I have a usb dongle ASUS BT400 bluetooth device
I downloaded the driver from Asus website and converted to BCM20702A1-0b05-17bc.hcd
I put it in lib/firmware/brcm
When I exec the command sudo dmesg | grep blue" is answer... "Direct firmware load for brcm/BCM20702A1-0b05-17bc.hcd failed with error - 2"

Can anyone help me?

winterheart, 2019-12-11:
Are you sure that BCM20702A1-0b05-17**bc**.hcd is requested? Asus BT400 should be BCM20702A1-0b05-17**cb**.hcd.
Please post info from commads:

```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb 
```

vpomax, 2019-12-12:
excuse me... I write wrong file it is CM20702A1-0b05-17cb.hcd.

root@proxmox:~# uname -a
Linux proxmox 5.0.15-1-pve #1 SMP PVE 5.0.15-1 (Wed, 03 Jul 2019 10:51:57 +0200) x86_64 GNU/Linux

root@proxmox:~# dmesg | grep -i Bluetooth
[    5.047151] Bluetooth: Core ver 2.22
[    5.047164] Bluetooth: HCI device and connection manager initialized
[    5.047166] Bluetooth: HCI socket layer initialized
[    5.047168] Bluetooth: L2CAP socket layer initialized
[    5.047170] Bluetooth: SCO socket layer initialized
[    5.189654] Bluetooth: hci0: BCM: chip id 63
[    5.190650] Bluetooth: hci0: BCM: features 0x07
[    5.206654] Bluetooth: hci0: proxmox
[    5.207646] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    5.214276] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0b05-17cb.hcd failed with error -2
[    5.214279] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0b05-17cb.hcd not found
[    5.983081] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.983081] Bluetooth: BNEP filters: protocol multicast
[    5.983110] Bluetooth: BNEP socket layer initialized
[   10.669393] Bluetooth: hci0: urb 000000001d16a0b5 failed to resubmit (2)

root@proxmox:~# lspci -nnvv | grep -A12 Broadcom
root@proxmox:~# 

root@proxmox:~# hciconfig -a
root@proxmox:~# 

root@proxmox:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04cf:0022 Myson Century, Inc. OCZ Alchemy Series Elixir II Keyboard
Bus 001 Device 003: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

winterheart, 2019-12-12:
Device now correctly loads firmware, but subsystem still can't up hci interface. "Bluetooth: hci0: urb 00000000ebcab887 failed to resubmit (2)" produced by usb_submit_urb() in kernel: https://www.kernel.org/doc/html/v4.12/driver-api/usb/error-codes.html . I don't quite follow what means error code 2.
Assuming that you generated *.hcd by own, can you please test https://github.com/winterheart/broadcom-bt-firmware/blob/master/brcm/BCM20702A1-0b05-17cb.hcd from latest version?

vpomax, 2019-12-12:
OK, I downloaded yor file and I've copied it in lib/firmware/brcm
restarted proxmox ... but I had similar error...

```
root@proxmox:~# dmesg | grep -i Bluetooth
[    2.818575] usb 1-1: Product: BCM920702 Bluetooth 4.0
[    5.385159] Bluetooth: Core ver 2.22
[    5.385173] Bluetooth: HCI device and connection manager initialized
[    5.385176] Bluetooth: HCI socket layer initialized
[    5.385179] Bluetooth: L2CAP socket layer initialized
[    5.385181] Bluetooth: SCO socket layer initialized
[    5.517511] Bluetooth: hci0: BCM: chip id 63
[    5.518505] Bluetooth: hci0: BCM: features 0x07
[    5.534510] Bluetooth: hci0: Broadcom Bluetooth Device
[    5.535499] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[    6.297502] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[    6.313662] Bluetooth: hci0: Broadcom Bluetooth Device
[    6.351707] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.351708] Bluetooth: BNEP filters: protocol multicast
[    6.351711] Bluetooth: BNEP socket layer initialized
[   11.137140] Bluetooth: hci0: urb 00000000fbb0e925 failed to resubmit (2)
```

winterheart, 2019-12-16:
That strange. I have same device, and mine's works well. Can you retest device on another OS, like recent Ubuntu or Fedora? Maybe your Linux kernel instance has issues.

vpomax, 2020-01-01:
I do not solved my problem but I understand the problem:
1) reinstalled proxmox from zero
2) apt-get update
3) apt-get install bluez
4) I installed my domotic system Hassio (Home Assistant) in a VM  using this script:
`wget -qO - https://raw.githubusercontent.com/whiskerz007/proxmox_hassos_install/master/install.sh | bash -s local-lvm`
5) hciconfig -a
```
root@nuc:~# hciconfig -ahci0:   Type: Primary  Bus: USB
        BD Address: 5C:F3:70:97:1C:F3  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING 
        RX bytes:4203 acl:0 sco:0 events:243 errors:0
        TX bytes:37769 acl:0 sco:0 commands:242 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH SNIFF 
        Link mode: SLAVE ACCEPT 
        Name: 'nuc'
        Class: 0x000104
        Service Classes: Unspecified
        Device Class: Computer, Desktop workstation
        HCI Version: 4.0 (0x6)  Revision: 0x15bb
        LMP Version: 4.0 (0x6)  Subversion: 0x220e
        Manufacturer: Broadcom Corporation (15)
```

My ASUS BT400 usb dongle is correctly recognized

6) I associated BT400 to my VM as described here (https://pve.proxmox.com/wiki/USB_Devices_in_Virtual_Machines)
`qm set 100 -usb0 host=0b05:17cb`

after few seconds it show this message:
`hciconfig -s[4081.788068] Bluetooth: hci0: urb 00000000a24bb708 failed to resubmit (2)`

If I launch the command...
```
root@nuc:~# dmesg | grep -i Bluetooth
[    2.791371] usb 1-1: Product: BCM920702 Bluetooth 4.0
[    5.403528] Bluetooth: Core ver 2.22
[    5.403543] Bluetooth: HCI device and connection manager initialized
[    5.403545] Bluetooth: HCI socket layer initialized
[    5.403547] Bluetooth: L2CAP socket layer initialized
[    5.403549] Bluetooth: SCO socket layer initialized
[    5.549589] Bluetooth: hci0: BCM: chip id 63
[    5.550587] Bluetooth: hci0: BCM: features 0x07
[    5.566596] Bluetooth: hci0: nuc
[    5.567589] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[    6.329616] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[    6.345617] Bluetooth: hci0: Broadcom Bluetooth Device
[    7.508709] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.508711] Bluetooth: BNEP filters: protocol multicast
[    7.508714] Bluetooth: BNEP socket layer initialized
[ 4081.788068] Bluetooth: hci0: urb 00000000a24bb708 failed to resubmit (2)
```
All works fine until I assign bluetooth devices to usb port
Can you help me to solve this problem?

winterheart, 2020-01-22:
Well, seems this not related to firmware issues. Not all USB devices can be forwarded into virtual machine (at least on technology which Proxmox based on), so this issue of virtual environment.

closed 2020-01-22 by winterheart