winterheart/broadcom-bt-firmware #19: Broadcom BCM94360cs - Bluetooth not working

kinggx9, 2020-01-24:
I am using a Broadcom BCM94360cs card
The connection to the phone works well
Connecting to the wireless headset has an error
My English is not good
Thanks

winterheart, 2020-01-24:
Hello.
Can you please post info about your system:
```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb 
```
Seems BCM94360cs is not related to firmware issues of BCM20702 etc, but still, let's find this out.

kinggx9, 2020-01-27:
uname -a:
Linux manjaro 4.19.97-1-lts #1 SMP Fri, 17 Jan 2020 23:09:48 +0000 x86_64 GNU/Linux
dmesg | grep -i Bluetooth:
Cannot ...
lspci -nnvv | grep -A12 Broadcom:
02:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
	Subsystem: Apple Inc. BCM4360 802.11ac Wireless Network Adapter [106b:0112]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 19
	Region 0: Memory at e1200000 (64-bit, non-prefetchable) [size=32K]
	Region 2: Memory at e1000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
	Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
hciconfig -a:
hci0:	Type: Primary  Bus: USB
	BD Address: 3C:15:C2:D5:52:D3  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:1486 acl:0 sco:0 events:72 errors:0
	TX bytes:1091 acl:0 sco:0 commands:72 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
lsusb:
Bus 001 Device 004: ID 0bda:881a Realtek Semiconductor Corp.

winterheart, 2020-01-27:
OK, seems this is our client. You have combined WiFi + Bluetooth device based on BCM 4360 chipset. Strange enough that no traces of Bluetooth stack at all.

Still, you need properly initialize WiFi device with own firmware NVRAM as described [here](https://github.com/winterheart/broadcom-bt-firmware/issues/3#issuecomment-318512097) and [here](https://wireless.wiki.kernel.org/en/users/drivers/brcm80211#broadcom_brcmfmac_driver). After that please reboot machine and re-post new output from my commands (especially `dmesg` one).

winterheart, 2020-01-27:
And additional [info](https://unix.stackexchange.com/questions/381848/installing-wifi-drivers-for-14e443a0-rev-3-broadcom-pcie-device-cant-find-bc) (mostly related to Ubuntu, but you can use it as base for your Manjaro/Arch distribution).

kinggx9, 2020-01-30:
Thank you
I have an error on the headset
Reason for missing pulseaudio-bluetooth component

closed 2020-01-30 by kinggx9