winterheart/broadcom-bt-firmware #22: 0489-e0a3.hcd

lazarkovacevic7, 2020-04-25:
Is it possible to have this driver? It is for Acer Aspire E15 model E5-575G-32PS.

Best regards
Lazar K.

winterheart, 2020-04-25:
Please post info about your system:
```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb
```

lazarkovacevic7, 2020-04-26:
> uname

 
`Linux Aspire-E5-575G 4.4.0-177-generic #207-Ubuntu SMP Mon Mar 16 01:16:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

> dmesg | grep -i Bluetooth

```
[    4.705667] Bluetooth: Core ver 2.21
[    4.705689] Bluetooth: HCI device and connection manager initialized
[    4.705814] Bluetooth: HCI socket layer initialized
[    4.705818] Bluetooth: L2CAP socket layer initialized
[    4.705826] Bluetooth: SCO socket layer initialized
[    5.454952] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.454956] Bluetooth: BNEP filters: protocol multicast
[    5.454961] Bluetooth: BNEP socket layer initialized
[    5.651308] Bluetooth: RFCOMM TTY layer initialized
[    5.651315] Bluetooth: RFCOMM socket layer initialized
[    5.651321] Bluetooth: RFCOMM ver 1.11
[   15.091034]  snd_seq_midi bluetooth snd_seq_midi_event snd_rawmidi snd_seq ecdh_generic snd_seq_device snd_timer snd input_leds rtsx_pci_ms soundcore ipmi_msghandler joydev memstick serio_raw idma64 shpchp virt_dma wmi mei_me mei intel_lpss_pci mac_hid intel_lpss binfmt_misc acpi_pad coretemp parport_pc ppdev lp parport cfg80211 autofs4 btrfs xor raid6_pq dm_mirror dm_region_hash dm_log rtsx_pci_sdmmc i915_bpo intel_ips i2c_algo_bit psmouse drm_kms_helper syscopyarea sysfillrect r8169 sysimgblt fb_sys_fops rtsx_pci mii ahci drm libahci fjes video
```

> lspci -nnvv | grep -A12 Broadcom

`//empty`

> hciconfig -a

```
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 90:CD:B6:07:3F:C2  ACL MTU: 1024:8  SCO MTU: 50:8
	UP RUNNING PSCAN ISCAN 
	RX bytes:831 acl:0 sco:0 events:70 errors:0
	TX bytes:4733 acl:0 sco:0 commands:69 errors:0
	Features: 0xff 0xfe 0x8f 0xfe 0xd8 0x3f 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'Aspire-E5-575G LK'
	Class: 0x3c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio
	Device Class: Computer, Laptop
	HCI Version: 4.1 (0x7)  Revision: 0x0
	LMP Version: 4.1 (0x7)  Subversion: 0x25a
	Manufacturer: Qualcomm (29)
```

> lsusb

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04f2:b571 Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 0489:e0a3 Foxconn / Hon Hai 
Bus 001 Device 002: ID 0bda:f179 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

winterheart, 2020-04-26:
This device is not related to Broadcom firmware issue. You need to ask on Ubuntu forums or bugzilla about issues with Bluetooth.

closed 2020-04-26 by winterheart