winterheart/broadcom-bt-firmware #23: BCM43142A0-0a5c-216d.hcd

s10gopal, 2020-04-29:
Bluetooth not able to scan device
`[    2.971302] Bluetooth: Core ver 2.22
[    2.971315] Bluetooth: HCI device and connection manager initialized
[    2.971318] Bluetooth: HCI socket layer initialized
[    2.971319] Bluetooth: L2CAP socket layer initialized
[    2.971322] Bluetooth: SCO socket layer initialized
[    3.173716] Bluetooth: hci0: BCM: chip id 70
[    3.174715] Bluetooth: hci0: BCM: features 0x06
[    3.190724] Bluetooth: hci0: BCM43142A
[    3.191715] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[    4.617145] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.617147] Bluetooth: BNEP filters: protocol multicast
[    4.617151] Bluetooth: BNEP socket layer initialized
[    5.264177] Bluetooth: hci0: command 0x0a0a tx timeout
[   13.392056] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[   15.664055] Bluetooth: hci0: command 0x1001 tx timeout
[   23.888015] Bluetooth: hci0: BCM: Reading local version info failed (-110)`

winterheart, 2020-04-30:
Hello. 

Seems your copy of BCM43142A0-0a5c-216d.hcd is corrupted. Try to download again and place it into /lib/firmware/brcm. If it won't help, please post output of these commands:
```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb
```

s10gopal, 2020-05-01:
HI,

uname -a
`Linux gopal-HP-Notebook 5.4.0-28-generic #32-Ubuntu SMP Wed Apr 22 17:40:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
`

dmesg | grep -i Bluetooth
```[    3.127631] Bluetooth: Core ver 2.22
[    3.127647] Bluetooth: HCI device and connection manager initialized
[    3.127651] Bluetooth: HCI socket layer initialized
[    3.127653] Bluetooth: L2CAP socket layer initialized
[    3.127657] Bluetooth: SCO socket layer initialized
[    3.361845] Bluetooth: hci0: BCM: chip id 70
[    3.362845] Bluetooth: hci0: BCM: features 0x06
[    3.378852] Bluetooth: hci0: BCM43142A
[    3.379853] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[    4.715004] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.715005] Bluetooth: BNEP filters: protocol multicast
[    4.715009] Bluetooth: BNEP socket layer initialized
[    5.455723] Bluetooth: hci0: command 0x0a0a tx timeout
[   13.647774] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[   15.983980] Bluetooth: hci0: command 0x1001 tx timeout
[   24.143805] Bluetooth: hci0: BCM: Reading local version info failed (-110)
```

lspci -nnvv | grep -A12 Broadcom
```03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43142 802.11b/g/n [14e4:4365] (rev 01)
	Subsystem: Hewlett-Packard Company BCM43142 802.11b/g/n [103c:804a]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at b1100000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: <access denied>
	Kernel driver in use: wl
	Kernel modules: bcma, wl
```

hciconfig -a
```hci0:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:322 acl:0 sco:0 events:8 errors:0
	TX bytes:40 acl:0 sco:0 commands:10 errors:0
	Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	Packet type: DM1 DH1 HV1 
	Link policy: 
	Link mode: SLAVE ACCEPT
```

lsusb
```Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b56c Chicony Electronics Co., Ltd HP TrueVision HD
Bus 001 Device 002: ID 0a5c:216d Broadcom Corp. BCM43142A0 Bluetooth 4.0
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mateuszkwiatkowski, 2020-05-12:
I see the same issue on Ubuntu 20.04 with Broadcom BCM20702A0. Here're outputs:
```
root@darkstar:/lib/firmware/brcm# uname -a
Linux darkstar 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
root@darkstar:/lib/firmware/brcm# dmesg | grep -i Bluetooth
[    9.271999] Bluetooth: Core ver 2.22
[    9.272017] Bluetooth: HCI device and connection manager initialized
[    9.272020] Bluetooth: HCI socket layer initialized
[    9.272022] Bluetooth: L2CAP socket layer initialized
[    9.272025] Bluetooth: SCO socket layer initialized
[    9.406051] Bluetooth: hci0: BCM: chip id 63
[    9.409069] Bluetooth: hci0: BCM: features 0x07
[    9.430173] Bluetooth: hci0: darkstar
[    9.432027] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[   11.373272] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.373273] Bluetooth: BNEP filters: protocol multicast
[   11.373278] Bluetooth: BNEP socket layer initialized
[   11.525471] Bluetooth: hci0: command 0x0a0a tx timeout
[   19.589487] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[   21.833481] Bluetooth: hci0: command 0x1001 tx timeout
[   29.829454] Bluetooth: hci0: BCM: Reading local version info failed (-110)
root@darkstar:/lib/firmware/brcm# lspci -nnvv | grep -A12 Broadcom
	DeviceName: Broadcom 5762
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin C routed to IRQ 31
	Bus: primary=15, secondary=16, subordinate=1e, sec-latency=0
	I/O behind bridge: 0000f000-0000ffff [size=4K]
	Memory behind bridge: fe600000-fe6fffff [size=1M]
	Prefetchable memory behind bridge: [disabled]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16+ MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
root@darkstar:/lib/firmware/brcm# hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:322 acl:0 sco:0 events:8 errors:0
	TX bytes:40 acl:0 sco:0 commands:10 errors:0
	Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	Packet type: DM1 DH1 HV1 
	Link policy: 
	Link mode: SLAVE ACCEPT 

root@darkstar:/lib/firmware/brcm# lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0b05:17ba ASUSTek Computer, Inc. N10 Nano 802.11n Network Adapter [Realtek RTL8192CU]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0835:1411 Action Star Enterprise Co., Ltd 
Bus 001 Device 007: ID 0424:274c Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 005: ID 0424:2734 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 004: ID 045e:00db Microsoft Corp. Natural Ergonomic Keyboard 4000 V1.0
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

winterheart, 2020-05-12:
```
[    9.432027] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
```
means that firmware silently refused (there should be additional string indicated different build version).

Only common things I see is Linux Kernel 5.4.0-29-generic. @mateuszkwiatkowski @s10gopal , which distributive do you have, what version of Ubuntu? I have Asus BT-400 (0b05:17cb) USB dongle, so I can try to reproduce on that version.

mateuszkwiatkowski, 2020-05-12:
Hi @winterheart ,
Many thanks for quick response. I have Ubuntu 20.04. It turned out that my firmware was indeed corrupted - I accidentally downloaded HTML from GitHub. After I fixed it with raw download something totally unexpected happened.
System was booting for few minutes (usually it's <1 minute) and after it booted the USB device is not listed by lsusb anymore and I see USB errors in dmesg:
```
[   16.646458] usb 5-1: device descriptor read/64, error -110
[   32.262463] usb 5-1: device descriptor read/64, error -110
[   32.498425] usb 5-1: new full-speed USB device number 3 using xhci_hcd
[   47.878456] usb 5-1: device descriptor read/64, error -110
[   63.494461] usb 5-1: device descriptor read/64, error -110
[   63.602503] usb usb5-port1: attempt power cycle
[   64.254165] usb 5-1: new full-speed USB device number 4 using xhci_hcd
[   69.510260] usb 5-1: device descriptor read/8, error -110
[   74.886259] usb 5-1: device descriptor read/8, error -110
[   75.122598] usb 5-1: new full-speed USB device number 5 using xhci_hcd
[   80.262253] usb 5-1: device descriptor read/8, error -110
[   85.638257] usb 5-1: device descriptor read/8, error -110
[   85.746453] usb usb5-port1: unable to enumerate USB device
[  143.766150] usb 5-1: new full-speed USB device number 6 using xhci_hcd
[  159.238439] usb 5-1: device descriptor read/64, error -110
[  174.854203] usb 5-1: device descriptor read/64, error -110
[  175.094159] usb 5-1: new full-speed USB device number 7 using xhci_hcd
[  190.474199] usb 5-1: device descriptor read/64, error -110
[  206.094208] usb 5-1: device descriptor read/64, error -110
[  206.202213] usb usb5-port1: attempt power cycle
[  206.858172] usb 5-1: new full-speed USB device number 8 using xhci_hcd
[  212.102508] usb 5-1: device descriptor read/8, error -110
[  217.478491] usb 5-1: device descriptor read/8, error -110
[  217.718149] usb 5-1: new full-speed USB device number 9 using xhci_hcd
[  222.854216] usb 5-1: device descriptor read/8, error -110
[  228.230218] usb 5-1: device descriptor read/8, error -110
[  228.338213] usb usb5-port1: unable to enumerate USB device
```

What's more surprising is the fact that after removing the firmware file the device didn't come back. I had to remove it from computer and insert it again. I'm lost now. :-)

winterheart, 2020-05-13:
@mateuszkwiatkowski seems there not enough power on USB. Can you try to use another port or temporary disable USB 3.0? This error also may occur if you connect device via bad cable or external hub.

mateuszkwiatkowski, 2020-05-14:
@winterheart after putting dongle into USB 2.0 port it started without errors in dmesg! Thank you very much!

winterheart, 2020-05-16:
@s10gopal can you please post `md5sum /lib/firmware/brcm/BCM43142A0-0a5c-216d.hcd` output? Should be:
```
77666fbfa599aa153406b2201ce4feba  /lib/firmware/brcm/BCM43142A0-0a5c-216d.hcd
```

s10gopal, 2020-05-17:
My apologies for the delayed response.

It is not a distro specific bug. I tried Ubuntu 18.04+ and Debian. 
`2290e2674fabe6ad99ac540b8561909b  /lib/firmware/brcm/BCM43142A0-0a5c-216d.hcd
`

s10gopal, 2020-05-17:
> Hi @winterheart ,
> Many thanks for quick response. I have Ubuntu 20.04. It turned out that my firmware was indeed corrupted - I accidentally downloaded HTML from GitHub. After I fixed it with raw [...]

After Downloading the file manually, i am also getting the same error. I don't think it is power problem in my case. I am using hp laptop and it is inbuilt and wifi is working properly.

s10gopal, 2020-05-17:
dmesg | grep -i Bluetooth
`[ 1574.141468] Bluetooth: HCI device and connection manager initialized
[ 1574.141471] Bluetooth: HCI socket layer initialized
[ 1574.141473] Bluetooth: L2CAP socket layer initialized
[ 1574.141476] Bluetooth: SCO socket layer initialized `

 lspci -nnvv | grep -A12 Broadcom
`03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43142 802.11b/g/n [14e4:4365] (rev 01)
	Subsystem: Hewlett-Packard Company BCM43142 802.11b/g/n [103c:804a]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at b1100000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: <access denied>
	Kernel driver in use: wl
	Kernel modules: bcma, wl`

hciconfig -a
`Empty`

lsusb
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f2:b56c Chicony Electronics Co., Ltd HP TrueVision HD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

dmesg [https://paste.ubuntu.com/p/QPPmTj3dr9/](url)

GUI show no Bluetooth dongle found.

winterheart, 2020-05-17:
Well, I have only few suggestions.

1. Try to power off / power on system (not just reboot)
2. Try to disable Secure Boot on BIOS/UEFI - there may be restrictions that disables USB capabilities.
3. Some reports indicates that need to enable IOMMU support in BIOS/UEFI.

Generally speaking, try to enable/disable USB-related options in BIOS/UEFI. If it allows, try to force enable USB 2.0 over 3.0.