winterheart/broadcom-bt-firmware #27: lubuntu 20.04 HP stream x360 11p-099nf bluetooth

lapinvb, 2020-09-08:
Fresh Lubuntu 20.04 installation on HP Stream x360 11p-099nf computer (with proprietary wifi drivers installed)
BlueDevil Assistant find no device to connect with.
After installation the bluetooth driver from https://github.com/winterheart/broadcom-bt-firmware/blob/master/brcm/BCM43142A0-0a5c-216c.hcd , bluetooth assistant find no device

linux vivien-hpstreamx360convertiblepc11 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

[    7.304166] Bluetooth: Core ver 2.22
[    7.304200] Bluetooth: HCI device and connection manager initialized
[    7.304207] Bluetooth: HCI socket layer initialized
[    7.304211] Bluetooth: L2CAP socket layer initialized
[    7.304217] Bluetooth: SCO socket layer initialized
[    7.411038] Bluetooth: HCI UART driver ver 2.3
[    7.411042] Bluetooth: HCI UART protocol H4 registered
[    7.411043] Bluetooth: HCI UART protocol BCSP registered
[    7.411067] Bluetooth: HCI UART protocol LL registered
[    7.411068] Bluetooth: HCI UART protocol ATH3K registered
[    7.411083] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    7.411149] Bluetooth: HCI UART protocol Intel registered
[    7.411218] Bluetooth: HCI UART protocol Broadcom registered
[    7.411234] Bluetooth: HCI UART protocol QCA registered
[    7.411235] Bluetooth: HCI UART protocol AG6XX registered
[    7.411250] Bluetooth: HCI UART protocol Marvell registered
[    8.106451] Bluetooth: hci0: BCM: chip id 70
[    8.107364] Bluetooth: hci0: BCM: features 0x06
[    8.123360] Bluetooth: hci0: BCM43142A
[    8.124362] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[   10.218448] Bluetooth: hci0: command 0x0a0a tx timeout
[   12.147094] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.147098] Bluetooth: BNEP filters: protocol multicast
[   12.149525] Bluetooth: BNEP socket layer initialized
[   18.218504] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[   20.522428] Bluetooth: hci0: command 0x1001 tx timeout
[   28.714497] Bluetooth: hci0: BCM: Reading local version info failed (-110)

02:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43142 802.11b/g/n [14e4:4365] (rev 01)
        Subsystem: Hewlett-Packard Company BCM43142 802.11b/g/n [103c:2230]
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 17
        Region 0: Memory at 90700000 (64-bit, non-prefetchable) [size=32K]
        Capabilities: <access denied>
        Kernel driver in use: wl
        Kernel modules: wl

03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader [10ec:5227] (rev 01)
        Subsystem: Hewlett-Packard Company RTS5227 PCI Express Card Reader [103c:802b]

hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:322 acl:0 sco:0 events:8 errors:0
        TX bytes:40 acl:0 sco:0 commands:10 errors:0
        Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1 
        Link policy: 
        Link mode: SLAVE ACCEPT 

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0a5c:216c Broadcom Corp. BCM43142A0 Bluetooth Device
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 05c8:036e Cheng Uei Precision Industry Co., Ltd (Foxlink) Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

winterheart, 2020-09-08:
Hello.

Seems you have combined Wifi+Bluetooth device, please check this section of README: https://github.com/winterheart/broadcom-bt-firmware#notes-about-combined-wifibluetooth-devices

lapinvb, 2020-09-10:
Thanks

wget https://github.com/khadas/android_hardware_amlogic_wifi/raw/b6709758755568e4a0ff6e80993be0fc64c77fb9/bcm_ampak/config/6255/nvram.txt
sudo cp nvram.txt /lib/firmware/brcm/brcmfmac43455-sdio.txt

but it still don't work

dmesg | grep -i Bluetooth
[    7.355964] Bluetooth: Core ver 2.22
[    7.355995] Bluetooth: HCI device and connection manager initialized
[    7.356002] Bluetooth: HCI socket layer initialized
[    7.356005] Bluetooth: L2CAP socket layer initialized
[    7.356012] Bluetooth: SCO socket layer initialized
[    7.460292] Bluetooth: HCI UART driver ver 2.3
[    7.460295] Bluetooth: HCI UART protocol H4 registered
[    7.460296] Bluetooth: HCI UART protocol BCSP registered
[    7.460317] Bluetooth: HCI UART protocol LL registered
[    7.460319] Bluetooth: HCI UART protocol ATH3K registered
[    7.460334] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    7.460403] Bluetooth: HCI UART protocol Intel registered
[    7.460471] Bluetooth: HCI UART protocol Broadcom registered
[    7.460486] Bluetooth: HCI UART protocol QCA registered
[    7.460487] Bluetooth: HCI UART protocol AG6XX registered
[    7.460503] Bluetooth: HCI UART protocol Marvell registered
[    8.172279] Bluetooth: hci0: BCM: chip id 70
[    8.173291] Bluetooth: hci0: BCM: features 0x06
[    8.189295] Bluetooth: hci0: BCM43142A
[    8.190299] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[   10.281794] Bluetooth: hci0: command 0x0a0a tx timeout
[   12.177143] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.177146] Bluetooth: BNEP filters: protocol multicast
[   12.177153] Bluetooth: BNEP socket layer initialized
**[   18.473853] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)**
[   20.777847] Bluetooth: hci0: command 0x1001 tx timeout
[   28.969836] Bluetooth: hci0: BCM: Reading local version info failed (-110)

winterheart, 2020-09-10:
Hello.
Have you tried copy own NVRAM from EFI? Like that:

```
Some new devices are storing the nvram which is needed in addition to the firmware by the driver in an EFI variable and the Windows driver can access it (this file should be optional in the case of PCIe devices). Currently brcmfmac does not support this automatically. First mount the efi vars into sysfs:

 mount -t efivarfs none /sys/firmware/efi/efivars

The content of the nvram is in this file:

/sys/firmware/efi/efivars/nvram-74b00bd9-805a-4d61-b51f-43268123d113

Copy this file where brcmfmac expects the nvram, for example:

 cat /sys/firmware/efi/efivars/nvram-74b00bd9-805a-4d61-b51f-43268123d113 > /lib/firmware/brcm/brcmfmac43241b4-sdio.txt
```
Please attach whole `dmesg` output to see what happens with wifi driver.

lapinvb, 2020-09-10:
# sudo mount -t efivarfs none /sys/firmware/efi/efivars
mount: /sys/firmware/efi/efivars: none déjà monté sur /sys/fs/bpf
# sudo cat /sys/firmware/efi/efivars/nvram-74b00bd9-805a-4d61-b51f-43268123d113 > /lib/firmware/brcm/brcmfmac43241b4-sdio.txt
bash: /lib/firmware/brcm/brcmfmac43241b4-sdio.txt: Permission non accordée

It seems I got any file nvram* in /sys/firmware/efi/efivars

[dmseg.txt](https://github.com/winterheart/broadcom-bt-firmware/files/5200494/dmseg.txt)

winterheart, 2020-09-10:
OK, seems firmware loading possible only with `brcmsmac` or `brcmfmac` driver. Propietary `wl` don't provide such interface, so you cannot use Bluetooth device with `wl`.

You can try one of `brcmsmac` or `brcmfmac`, but I can't tell if there any support of your WiFi device. If you'll get working wifi with `brcmXmac` family driver, it's likely you'll get support Bluetooth too. With `wl` you can work only with wifi without Bluetooth capabilities.

lapinvb, 2020-09-10:
Clear. Thanks a lot. How to change wl to brcmXmac , and go back if not working ?

winterheart, 2020-09-11:
Sorry, there I cannot provide any advices, since I don't have such devices to research.