winterheart/broadcom-bt-firmware #28: Insignia BCM20702B0 on latest kernels

parkerlreed, 2020-09-25:
**Describe the bug**

Insignia BCM20702B0 seems to have stopped working as of the latest kernels

Any attempts to reset fail

```
$ sudo hciconfig hci0 reset
Can't init device hci0: Connection timed out (110)
```

**Provide info about your system**
```
Linux wolfcola 5.8.10-arch1-1 #1 SMP PREEMPT Thu, 17 Sep 2020 18:01:06 +0000 x86_64 GNU/Linux
```
```
[73733.114714] Bluetooth: Core ver 2.22
[73733.114733] Bluetooth: HCI device and connection manager initialized
[73733.114737] Bluetooth: HCI socket layer initialized
[73733.114738] Bluetooth: L2CAP socket layer initialized
[73733.114741] Bluetooth: SCO socket layer initialized
[73733.158701] audit: type=1130 audit(1600995058.457:101): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[73733.165567] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[73733.165568] Bluetooth: BNEP filters: protocol multicast
[73733.165571] Bluetooth: BNEP socket layer initialized
[73733.182653] audit: type=1130 audit(1600995058.480:104): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth-mesh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[73733.240399] Bluetooth: hci0: BCM: chip id 73
[73733.241369] Bluetooth: hci0: BCM: features 0x07
[73733.257400] Bluetooth: hci0: BCM20702B
[73733.257402] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[73733.260941] Bluetooth: hci0: BCM20702B0 'brcm/BCM20702B0-19ff-0239.hcd' Patch
[73735.361635] Bluetooth: hci0: command 0x0a0a tx timeout
[73743.468662] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[73743.468664] Bluetooth: hci0: BCM: Patch failed (-110)
[73745.815389] Bluetooth: hci0: command 0x1001 tx timeout
[73753.708856] Bluetooth: hci0: BCM: Reading local version info failed (-110)
```
```
hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN 
        RX bytes:322 acl:0 sco:0 events:8 errors:0
        TX bytes:40 acl:0 sco:0 commands:10 errors:0
        Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1 
        Link policy: 
        Link mode: SLAVE ACCEPT 
```
```
Bus 001 Device 005: ID 19ff:0239 Dynex Bluetooth 4.0 Adapter [Broadcom, 1.12, BCM20702A0]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         1 
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x19ff Dynex
  idProduct          0x0239 Bluetooth 4.0 Adapter [Broadcom, 1.12, BCM20702A0]
  bcdDevice            1.12
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00da
    bNumInterfaces          4
[... 253 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      1 Device Firmware Update
      bInterfaceProtocol      1 
      iInterface              0 
      Device Firmware Upgrade Interface Descriptor:
        bLength                             9
        bDescriptorType                    33
        bmAttributes                        5
          Will Not Detach
          Manifestation Tolerant
          Upload Unsupported
          Download Supported
        wDetachTimeout                   5000 milliseconds
        wTransferSize                      64 bytes
        bcdDFUVersion                   1.10
```

winterheart, 2020-09-27:
Please post output of `md5sum /lib/firmware/brcm/BCM20702B0-19ff-0239.hcd` command and check it with [DEVICES.md](https://github.com/winterheart/broadcom-bt-firmware/blob/master/DEVICES.md) reference. If sum does not match, then yours is broken, redownload firmware and reboot system.

Issue with -110 error also indicates issues with powering device - bad cable or low voltage on USB ports. Try also attach device into USB2 (not USB3) port.

parkerlreed, 2020-09-28:
Yeah that was on the front USB 2. I'll check the md5 and try a back port when I get back home. Thanks.

parkerlreed, 2020-09-28:
Was able to check the md5sum remotely. For some reason it did not match. Will update when I can get home and pug it back in.

parkerlreed, 2020-09-29:
```
[   25.481048] usb 1-10: new full-speed USB device number 5 using xhci_hcd
[   25.624667] usb 1-10: New USB device found, idVendor=19ff, idProduct=0239, bcdDevice= 1.12
[   25.624673] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   25.624677] usb 1-10: Product: BCM20702A0
[   25.624680] usb 1-10: Manufacturer: Broadcom Corp
[   25.624683] usb 1-10: SerialNumber: BC14EF9CDCC9
[   25.703922] Bluetooth: Core ver 2.22
[   25.703937] NET: Registered protocol family 31
[   25.703938] Bluetooth: HCI device and connection manager initialized
[   25.703941] Bluetooth: HCI socket layer initialized
[   25.703942] Bluetooth: L2CAP socket layer initialized
[   25.703944] Bluetooth: SCO socket layer initialized
[   25.716514] usbcore: registered new interface driver btusb
[   25.735462] audit: type=1130 audit(1601349817.571:119): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   25.746863] NET: Registered protocol family 38
[   25.751360] audit: type=1130 audit(1601349817.587:120): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   25.758738] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   25.758740] Bluetooth: BNEP filters: protocol multicast
[   25.758743] Bluetooth: BNEP socket layer initialized
[   25.763008] audit: type=1130 audit(1601349817.601:121): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth-mesh comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   25.828438] Bluetooth: hci0: BCM: chip id 73
[   25.829437] Bluetooth: hci0: BCM: features 0x07
[   25.845434] Bluetooth: hci0: BCM20702B
[   25.845438] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0000
[   25.847638] Bluetooth: hci0: BCM20702B0 'brcm/BCM20702B0-19ff-0239.hcd' Patch
[   26.580519] Bluetooth: hci0: BCM20702B0 Generic USB Detuned Class 1 @ 20 MHz - BDROID-TV
[   26.580527] Bluetooth: hci0: BCM20702B0 (002.001.014) build 0557
[   26.702290] Bluetooth: RFCOMM TTY layer initialized
[   26.702299] Bluetooth: RFCOMM socket layer initialized
[   26.702308] Bluetooth: RFCOMM ver 1.11
```

All seems well but still non functional.

```
[parker@wolfcola ~]$ sudo hciconfig dev
hci0:   Type: Primary  Bus: USB
        BD Address: BC:14:EF:9C:DC:C9  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING 
        RX bytes:2182 acl:0 sco:0 events:187 errors:0
        TX bytes:36367 acl:0 sco:0 commands:193 errors:0

[parker@wolfcola ~]$ sudo hciconfig hci0 reset
Can't down device hci0: Device or resource busy (16)
[parker@wolfcola ~]$ bluetoothctl 
Agent registered
[bluetooth]# devices
No default controller available
```

parkerlreed, 2020-09-29:
Odd part is scanning is working

```
[parker@wolfcola ~]$ sudo hcitool lescan
LE Scan ...
F4:BC:DA:30:C6:53 FD5A10B9A2BE
F4:BC:DA:30:C6:53 (unknown)
2C:26:17:16:53:63 (unknown)
2C:26:17:16:53:63 OMVR-V190
D7:09:EA:02:EE:DE (unknown)
D7:09:EA:02:EE:DE Tile
```

parkerlreed, 2020-09-29:
If this helps at all
```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2020-09-28 23:33:31 EDT; 12s ago
       Docs: man:bluetoothd(8)
   Main PID: 2102 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 19121)
     Memory: 896.0K
     CGroup: /system.slice/bluetooth.service
             └─2102 /usr/lib/bluetooth/bluetoothd

Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEScanIntervalConnect” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEScanWindowConnect” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEMinConnectionInterval” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEMaxConnectionInterval” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEConnectionLatency” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEConnectionSupervisionTimeout” in group “Controller”
Sep 28 23:33:31 wolfcola bluetoothd[2102]: src/main.c:parse_controller_config() Key file does not have key “LEAutoconnecttimeout” in group “Controller”
Sep 28 23:33:31 wolfcola systemd[1]: Started Bluetooth service.
Sep 28 23:33:31 wolfcola bluetoothd[2102]: Starting SDP server
Sep 28 23:33:31 wolfcola bluetoothd[2102]: Bluetooth management interface 1.17 initialized
```

parkerlreed, 2020-10-15:
And it's randomly working again... Jeeze.

closed 2020-10-15 by parkerlreed