winterheart/broadcom-bt-firmware #30: Kernel 5.12 - Bluetooth: hci0: BCM: firmware Patch file not found

Xavron, 2021-05-26:
**Describe the bug**
Resume computer and screen may wake up to this error. Never happened before kernel 5.12.6.

uname -a
Linux 5.12.6-200.fc33.x86_64

dmesg | grep -i Bluetooth
usb 1-9: Product: BCM920702 Bluetooth 4.0
Core ver 2.22
...
initialized...
...
hci0: BCM20702A1 (001.002.014) build 1764
hci0: BCM: firmware Patch file not found, tried:
hci0: BCM: 'brcm/BCM20702A1-0a5c-21e8.hcd'
hci0: BCM: 'brcm/BCM-0a5c-21e8.hcd'
...

Xavron, 2021-05-26:
I'm not sure what's going on. Maybe Redhat messed up Fedora 33 or something but the hcd file wasn't in /lib/... and dnf isn't finding broadcom-bt-firmware even though I thought that I installed it from there and another update from a different repo isn't being found either.

If it looks like something to you, let me know. I will see if the hcd file fixes it. Thanks.

Xavron, 2021-05-27:
Adding the file produces:
hci0: BCM20702A1 (001.002.014) build 0000
hci0: BCM20702A1 'brcm/BCM20702A1-0a5c-21e8.hcd' Patch
hci0: Broadcom Bluetooth Device

Guess its fine now but I still have no idea what happened as it was always fine before now.

Xavron, 2021-05-27:
Finally figured out why broadcom-bt-firmware is missing with rpmfusion nonfree and dnf... the rpmfusion devs decided to change how rpmfusion repo's worked breaking existing uses.

Then they or Redhat have that messed up and put broadcom-bt-firmware in nonfree tainted repo which isn't even available on Fedora 33 through dnf install. Only free tainted is.

Looks like have to manually download the rpm from rpmfusion non-free tainted folder at this time.

Xavron, 2021-05-27:
Reinstalling the rpm added additional 113 missing files and thus can solve the issue I suppose. Probably more important why files went missing in the first place.

Probably not a bug for here.

Xavron, 2021-05-28:
And now dnf has broadcom-bt-firmware again lol. Feels like I'm going nuts. It was there, it wasn't there, its there again. :-)

garcipat, 2022-11-13:
> Adding the file manually produces: hci0: BCM20702A1 (001.002.014) build 0000 hci0: BCM20702A1 'brcm/BCM20702A1-0a5c-21e8.hcd' Patch hci0: Broadcom Bluetooth Device
> 
> Guess its fine now but I still have no idea what happened as it was always fine before now.
> 
> Also dnf history somewhat shows broadcom-bt-firmware was installed previously. It doesn't display the package name. So I suppose it changed with rpmfusion since then. Either that or dnf is totally messed up.
> 
> I will leave it to you to close. Maybe you will see something I don't see.

What file did you add? Im getting the same error (among others) with my bluetooth. it works for e second and is disbled. When query "dmesg | grep Bluetooth" Im getting that same error. Just installed Ubuntu fresh and getting this.

berrycow2, 2023-06-11:
> > Adding the file manually produces: hci0: BCM20702A1 (001.002.014) build 0000 hci0: BCM20702A1 'brcm/BCM20702A1-0a5c-21e8.hcd' Patch hci0: Broadcom Bluetooth Device
> > Guess its fine now but I still have no idea what happened as it was always fine before now.
> > Also dnf history somewhat shows broadcom-bt-firmware was installed previously. It doesn't display the package name. So I suppose it changed with rpmfusion since then. Either that or dnf is totally messed up.
> > I will leave it to you to close. Maybe you will see something I don't see.
> 
> What file did you add? Im getting the same error (among others) with my bluetooth. it works for e second and is disbled. When query "dmesg | grep Bluetooth" Im getting that same error. Just installed Ubuntu fresh and getting this.
> 
> Edit: okay I just installed the latest .deb and it seems to solve the problem after a restart.

Can you advise that deb file? Got same problem on my Lenovo

berrycow2, 2023-06-11:
It's ok, found it here: https://github.com/winterheart/broadcom-bt-firmware/releases

gwapocom, 2024-08-08:
> Adding the file manually produces: hci0: BCM20702A1 (001.002.014) build 0000 hci0: BCM20702A1 'brcm/BCM20702A1-0a5c-21e8.hcd' Patch hci0: Broadcom Bluetooth Device
>
> Guess its fine now but I still [...]

How to add that file where did you add it???

winterheart, 2024-08-12:
> Hi, same here
> 
> ```
> [   10.858572] Bluetooth: hci0: BCM: chip id 84
> [   10.859131] Bluetooth: hci0: BCM: features 0x0f
> [   10.860321] Bluetooth: hci0: BCM4324B3
> [   10.860334] Bluetooth: hci0: BCM4324B3 (002.004.006) build 0000
> [   10.863195] bluetooth hci0: firmware: failed to load brcm/BCM4324B3.hcd (-2)
> [   10.863456] bluetooth hci0: firmware: failed to load brcm/BCM4324B3.hcd (-2)
> [   10.863518] bluetooth hci0: firmware: failed to load brcm/BCM.hcd (-2)
> [   10.863769] bluetooth hci0: firmware: failed to load brcm/BCM.hcd (-2)
> [   10.863781] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
> [   10.863833] Bluetooth: hci0: BCM: 'brcm/BCM4324B3.hcd'
> [   10.863867] Bluetooth: hci0: BCM: 'brcm/BCM.hcd'
> [   10.933392] rt5645 i2c-10EC5640:00: Device with ID register 0x6271 is not rt5645 or rt5650
> [   11.013913] rt5651 i2c-10EC5640:00: Device with ID register 0x6271 is not rt5651
> [   11.052245] rt5670 i2c-10EC5640:00: quirk GPIO1 is IRQ
> [   11.052257] rt5670 i2c-10EC5640:00: quirk DMIC enabled
> [   11.052262] rt5670 i2c-10EC5640:00: quirk DMIC2 on INR pin
> [   11.052266] rt5670 i2c-10EC5640:00: quirk JD mode 1
> ```
> 
> Hardware:
> 
> ```
> System Information
>         Manufacturer: Dell Inc.
>         Product Name: Venue 10 Pro 5055
>         Version: A08
>         Wake-up Type: Power Switch
>         SKU Number: Venue 10 Pro 5055
>         Family: Venue
> ```
> 
> Debian Trixie, Kernel Linux 6.10.3-amd64
> 
> Thanks.

Hello. Your device belongs to another type of Broadcom BT devices - [UART](https://www.catalog.update.microsoft.com/Search.aspx?q=Broadcom%20bluetooth%20%20uart), which is not covered by this  repository. You can try to manually download cab from MS Update, unpack and try one of HCD files inside on it.

winterheart, 2024-08-12:
You need one of BCM4324B3_002.004.006.*.hcd files.