winterheart/broadcom-bt-firmware #31: WM-BAC-BM28 module Bluetooth UART interfacing issue

Manikantag05, 2021-07-27:
I am trying to interface WM-BAC-BM28 module via UART to IMX6SX. Please provide me the device tree, GPIO configuration and other required details.

Thanks..

winterheart, 2021-07-27:
Please provide required information about your system:

```
uname -a
dmesg | grep -i Bluetooth
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb

Manikantag05, 2021-07-27:
Hi,
Please find the details below:

uname -a
Linux imx6sx-35som-debug 4.14.200+g52c76892ff6f #1 SMP PREEMPT Wed Oct 21 03:59:43 UTC 2020 armv7l GNU/Linux

dmesg | grep -i Bluetooth
[    0.306290] Bluetooth: Core ver 2.22
[    0.306355] Bluetooth: HCI device and connection manager initialized
[    0.306374] Bluetooth: HCI socket layer initialized
[    0.306387] Bluetooth: L2CAP socket layer initialized
[    0.306423] Bluetooth: SCO socket layer initialized
[    2.988885] Bluetooth: HCI UART driver ver 2.3
[    2.993400] Bluetooth: HCI UART protocol H4 registered
[    2.998547] Bluetooth: HCI UART protocol BCSP registered
[    3.003916] Bluetooth: HCI UART protocol ATH3K registered
[    3.009571] Bluetooth: HCI UART protocol Broadcom registered
[    4.378120] Bluetooth: RFCOMM TTY layer initialized
[    4.383133] Bluetooth: RFCOMM socket layer initialized
[    4.388303] Bluetooth: RFCOMM ver 1.11
[    4.392108] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.397427] Bluetooth: BNEP filters: protocol multicast
[    4.402689] Bluetooth: BNEP socket layer initialized
[    4.407664] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    4.413617] Bluetooth: HIDP socket layer initialized

lspci -nnvv | grep -A12 Broadcom 
> No output (Nothing printed)

lspci
> No output (Nothing printed)

hciconfig -a
> No output (Nothing printed)

hcidump 
HCI sniffer - Bluetooth packet analyzer ver 5.54
Can't attach to device hci0. No such device(19)

lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0002

Thanks..

winterheart, 2021-08-01:
Hello.

Sorry, can't help here. This is UART device, while broadcom-bt-firmware aimed to PCIe/USB SoC device family. As I can see from your output, there no firmware issues related to BCM chipsets.

closed 2021-08-01 by winterheart