winterheart/broadcom-bt-firmware #33: No more bluetooth

daboynb, 2022-06-24:
Hi, I'm on Debian 11. 
I've tried installing this because my Bluetooth wasn't working really good but, after installing the deb package it not work anymore.
I've tried:
- removing the driver from lib/firmware/brcm
- removing the directory and re-install all packages:  blueman firmware-brcm80211 firmware-misc-nonfree firmware-b43-installer 
Nothing seems to bring back the Bluetooth adapter.

sudo dmesg | grep -i blue
- [    7.829519] usb 1-4: Product: BCM920703 Bluetooth 4.1
- [   26.510579] Bluetooth: Core ver 2.22
- [   26.510599] Bluetooth: HCI device and connection manager initialized
- [   26.510602] Bluetooth: HCI socket layer initialized
- [   26.510604] Bluetooth: L2CAP socket layer initialized
- [   26.510607] Bluetooth: SCO socket layer initialized
- [   26.622929] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
- [   26.622931] Bluetooth: BNEP filters: protocol multicast
- [   26.622934] Bluetooth: BNEP socket layer initialized
- [   28.654778] Bluetooth: hci0: command 0x1001 tx timeout
- [   36.882694] Bluetooth: hci0: BCM: Reading local version info failed (-110)

sudo dmesg | grep -i firmware 

- `[    0.082797] Spectre V2 : Enabling Restricted Speculation for firmware calls
- [    0.285122] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
- [    7.245477] i915 0000:00:02.0: firmware: direct-loading firmware i915/skl_dmc_ver1_27.bin
- [    7.245762] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/skl_dmc_ver1_27.bin (v1.27)
- [   25.955065] platform regulatory.0: firmware: direct-loading firmware regulatory.db
- [   25.976270] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
- [   26.166382] brcmfmac 0000:02:00.0: firmware: direct-loading firmware brcm/brcmfmac43602-pcie.bin
- [   26.166579] brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac43602-pcie.Dell Inc.-XPS 15 9550.txt (-2)
- [   26.167990] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
- [   26.167992] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.Dell Inc.-XPS 15 9550.txt failed with error -2
- [   26.168019] brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac43602-pcie.txt (-2)
- [   26.168021] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
- [   26.689173] brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac43602-pcie.clm_blob (-2)
- [   26.689619] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/1 wl0: Nov 10 2015 06:38:10 version 7.35.177.61 (r598657) FWID 01-ea662a8c

winterheart, 2022-06-24:
Hello.
Seems you have a combined WiFi+Bluetooth device. You need updated firmware for WiFi part of device. Please see https://github.com/winterheart/broadcom-bt-firmware#notes-about-combined-wifibluetooth-devices section in README.md. Since you have Debian, you need install [firmware-brcm80211](https://packages.debian.org/firmware-brcm80211) in order to place BIN part of firmware. TXT part of firmware can be obtained from NVRAM as described here https://wireless.wiki.kernel.org/en/users/drivers/brcm80211#broadcom_brcmfmac_driver

winterheart, 2022-06-24:
There also some info that can be useful: https://github.com/winterheart/broadcom-bt-firmware/issues/3#issuecomment-318512097

daboynb, 2022-06-24:
Thanks for the reply, but none of this fixed the issue. WiFi is working normally and the firmware-brcm80211 is already installed.
Before installing the deb all was working, there's no method to revert? thanks.

daboynb, 2022-07-08:
Looking inside the /sys/firmware/efi/efivars... I don't have any nvram files.

closed 2022-07-27 by daboynb