winterheart/broadcom-bt-firmware #36: Kernel 5.15, Ubuntu 22.04.1 LTS, Broadcom (Original) BCM20702A0 - firmware update patch to BCM20702A1 - FAILED

LibUto, 2022-11-30:
**Describe the bug**

Kernel 5.15, Ubuntu 22.04.1 LTS, Broadcom (Original) BCM20702A0 - firmware update patch to BCM20702A1 - FAILED

**Provide info about your system**
uname -a

Linux jjfish 5.15.0-53-generic #59-Ubuntu SMP Mon Oct 17 18:53:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

dmesg | grep -i Bluetooth

 4.413182] Bluetooth: Core ver 2.22
[    4.413230] NET: Registered PF_BLUETOOTH protocol family
[    4.413256] Bluetooth: HCI device and connection manager initialized
[    4.413264] Bluetooth: HCI socket layer initialized
[    4.413267] Bluetooth: L2CAP socket layer initialized
[    4.413274] Bluetooth: SCO socket layer initialized
[    4.558425] Bluetooth: hci0: BCM: chip id 63
[    4.559413] Bluetooth: hci0: BCM: features 0x07
[    4.579373] Bluetooth: hci0: BCM20702A
[    4.579380] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    4.582180] Bluetooth: hci0: BCM20702A1 'brcm/BCM20702A1-0a5c-21e8.hcd' Patch
[    4.812111] usb 2-1.6.3: Product: Dell Wireless 365 Bluetooth Module
[    6.664073] Bluetooth: hci0: command 0x0a0a tx timeout
[    7.360480] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.360491] Bluetooth: BNEP filters: protocol multicast
[    7.360499] Bluetooth: BNEP socket layer initialized
[    8.811055] Bluetooth: RFCOMM TTY layer initialized
[    8.811072] Bluetooth: RFCOMM socket layer initialized
[    8.811081] Bluetooth: RFCOMM ver 1.11
[   14.695922] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[   14.695941] Bluetooth: hci0: BCM: Patch failed (-110)
[   16.903920] Bluetooth: hci0: command 0x1001 tx timeout
[   24.935909] Bluetooth: hci0: BCM: Reading local version info failed (-110)
[ 2151.728678] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.728688] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.728691] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.728693] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.728695] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729606] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729613] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729616] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729619] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729623] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729626] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729629] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.729632] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730611] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730623] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730628] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730633] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730637] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730640] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730644] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.730648] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731610] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731619] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731623] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731625] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731628] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731631] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731634] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.731637] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732601] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732610] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732614] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732618] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732622] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732625] Bluetooth: hci1: Received unexpected HCI Event 00000000
[ 2151.732630] Bluetooth: hci1: Received unexpected HCI Event 00000000
[... 177 lines cut ...]
	BD Address: 88:9F:FA:B3:8D:D5  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN INQUIRY 
	RX bytes:44497 acl:0 sco:0 events:646 errors:0
	TX bytes:7241 acl:0 sco:0 commands:262 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0x9b 0xff 0x79 0x83
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
Can't read local name on hci1: Input/output error (5)

lsusb
```
us 002 Device 006: ID 413c:8160 Dell Computer Corp. Wireless 365 Bluetooth
Bus 002 Device 005: ID 413c:8162 Dell Computer Corp. Integrated Touchpad [Synaptics]
Bus 002 Device 004: ID 413c:8161 Dell Computer Corp. Integrated Keyboard
Bus 002 Device 003: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0c45:6461 Microdia Laptop_Integrated_Webcam_1.3M
Bus 001 Device 007: ID 04ca:0061 Lite-On Technology Corp. USB Optical Mouse
Bus 001 Device 006: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
Bus 001 Device 003: ID 058f:9254 Alcor Micro Corp. Hub
Bus 001 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

winterheart, 2022-11-30:
Hello.
```
[ 14.695922] Bluetooth: hci0: BCM: Patch command 0a0a failed (-110)
[ 14.695941] Bluetooth: hci0: BCM: Patch failed (-110)
```
This may indicates that you have corrupted firmware. Please check md5sum of your fw file `/lib/firmware/brcm/BCM20702A1-0a5c-21e8.hcd`:

```
md5sum /lib/firmware/brcm/BCM20702A1-0a5c-21e8.hcd
3bdef9e78561f521efb303220982bdeb  /lib/firmware/brcm/BCM20702A1-0a5c-21e8.hcd
```

LibUto, 2022-12-01:
Even though I confirmed earlier that the cited patch worked yesterday, I found that its behavior is inconsistent and erratic, though working partially. Despite the presence of several Bluetooth devices, only one or two devices are getting recognized, and getting disconnected off and on.The Bluetooth ON icon is also not appearing on the task bar display. Even with the one or two devices working, the "patch failure" output is seen against the  "dmesg | grep -i Bluetooth" command

winterheart, 2022-12-01:
Another source of unstable working is underpowered USB connection or bad USB-cabling. Please check on another port, if dongle connected via cable, try connect directly to port. If connected to USB3, try to use USB2 port.

lorn10, 2023-12-26:
Hi all

I just stumbled by chance upon this topic and want to give the following hint. Try to disable the USB **auto suspend** feature:

`sudo nano /etc/default/grub`

Add the following value to the "GRUB_CMDLINE_LINUX_DEFAULT" line.

`usbcore.autosuspend=-1`

The result should then look like "GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.autosuspend=-1". After that update the GRUB config with:

`sudo update-grub`

This may help or not. I think those Broadcom BT 4.x controllers can have sometimes a quite special behavior. And those with Apple branding are sometimes even more weird (especially if the have an older firmware) . Mine is a BCM94360CSAX and not even listed with `lsusb`. Only the "BCM2046B1 USB 2.0 Hub" entry appears.

Whatever, - it should be also added that there can exist a an additional problem with the bluez BT stack. Which can be resolved by upgrading to a more recent version.

Source:
https://askubuntu.com/questions/1435112/ubuntu-20-04-bluetooth-mouse-is-disconnecting-and-reconnecting-every-minute-or/1498046#1498046
https://askubuntu.com/questions/1303731/how-to-change-bluetooth-timeout-settings-for-bluetooth-mouse/1362310#1362310

LibUto, 2023-12-26:
Hi,
Thanks for your update. I'll attempt the procedure suggested and assess the outcome.
Thanks and regards,
LibUto

https://askubuntu.com/questions/1435112/ubuntu-20-04-bluetooth-mouse-is-disconnecting-and-reconnecting-every-minute-or/1498046#1498046
https://askubuntu.com/questions/1303731/how-to-change-bluetooth-timeout-settings-for-bluetooth-mouse/1362310#1362310

danbates2, 2024-05-07:
I also have a BCM20702A0 I'd like to try update because of poor sound quality issues and after trying lots of other avenues.

These .hcd files here are for **BCM20702A1**? 
Is there a suitable file here?

I'm on Ubuntu 24.04 LTS
Kernel: Linux 6.8.0-31-generic

winterheart, 2024-05-07:
Check README.md on how to detect your device and required firmware for it.

danbates2, 2024-05-08:
I don't get the expected output at the first stage.

```
$ sudo dmesg | grep -i bluetooth

[    6.862113] Bluetooth: Core ver 2.22
[    6.862157] NET: Registered PF_BLUETOOTH protocol family
[    6.862160] Bluetooth: HCI device and connection manager initialized
[    6.862165] Bluetooth: HCI socket layer initialized
[    6.862168] Bluetooth: L2CAP socket layer initialized
[    6.862177] Bluetooth: SCO socket layer initialized
[    8.010732] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.010739] Bluetooth: BNEP filters: protocol multicast
[    8.010745] Bluetooth: BNEP socket layer initialized
[    8.015707] Bluetooth: MGMT ver 1.22
[   10.895177] Bluetooth: RFCOMM TTY layer initialized
[   10.895189] Bluetooth: RFCOMM socket layer initialized
[   10.895199] Bluetooth: RFCOMM ver 1.11
```

There's no information on the firmware version.

I've also tried installing the .deb package to no avail.

winterheart, 2024-05-08:
Please open separate issue and additional info about you hardware.

lorn10, 2024-05-09:
@danbates2 
If you are on a Mac / Apple hardware then the easiest way is to boot up Mac OS. For example, versions like 10.11 "El Capitan" or newer are automatically updating the BCM94360CSAX BT controller to the latest firmware. But when it comes to Linux I have unfortunately no idea if those special "Apple" Broadcom controllers are at all supported in `fwupd`.

Well, my BCM94360CSAX BT controller seems to be supported when I run `fwupdmgr get-devices` but I it's unclear from where I can get the corresponding firmware files. Probably there exist a way to grab / download them via `sysfs` and make an own "firmware library". :thinking: