winterheart/broadcom-bt-firmware #40: BCM43142A0-0a5c-21d7.hcd

9srp6, 2024-03-27:
I can't seem to get the bluetooth working even after installing the firmware for my chipset.
Bluetoothctl just shows that there is no default controller available.

`dmesg | grep -i Bluetooth`:
```
[   12.696489] Bluetooth: Core ver 2.22
[   12.696514] NET: Registered PF_BLUETOOTH protocol family
[   12.696517] Bluetooth: HCI device and connection manager initialized
[   12.696522] Bluetooth: HCI socket layer initialized
[   12.696524] Bluetooth: L2CAP socket layer initialized
[   12.696531] Bluetooth: SCO socket layer initialized
[   13.210723] Bluetooth: hci0: BCM: chip id 70
[   13.211721] Bluetooth: hci0: BCM: features 0x06
[   13.227728] Bluetooth: hci0: BCM43142A
[   13.227734] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[   13.798750] Bluetooth: hci0: BCM43142A0 'brcm/BCM43142A0-0a5c-21d7.hcd' Patch
[   15.951718] Bluetooth: hci0: command 0xfc4c tx timeout
[   24.058347] Bluetooth: hci0: BCM: Patch command fc4c failed (-110)
[   24.058364] Bluetooth: hci0: BCM: Patch failed (-110)
[   26.191581] Bluetooth: hci0: command 0x0c03 tx timeout
[   34.084768] Bluetooth: hci0: BCM: Reset failed (-110)
[  692.007749] Bluetooth: hci0: command 0x0c03 tx timeout
[  699.900453] Bluetooth: hci0: BCM: Reset failed (-110)
[ 1089.851701] Bluetooth: hci0: command 0x0c03 tx timeout
[ 1097.957361] Bluetooth: hci0: BCM: Reset failed (-110)
[ 1102.650218] Bluetooth: hci0: command 0x0c03 tx timeout
[ 1110.542497] Bluetooth: hci0: BCM: Reset failed (-110)
[21425.021347] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[21425.021351] Bluetooth: BNEP filters: protocol multicast
[21425.021355] Bluetooth: BNEP socket layer initialized
[65887.994553] Bluetooth: hci0: command 0x0c03 tx timeout
[65896.101077] Bluetooth: hci0: BCM: Reset failed (-110)
``` 
`uname -a`:
```
Linux SamServer 6.8.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 16 Mar 2024 17:15:35 +0000 x86_64 GNU/Linux
```
`lspci -nnvv | grep -A12 Broadcom`:
```
06:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43142 802.11b/g/n [14e4:4365] (rev 01)
        Subsystem: Dell Wireless 1704 802.11n + BT 4.0 [1028:0016]
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 18
        Region 0: Memory at ab200000 (64-bit, non-prefetchable) [size=32K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
        Capabilities: [58] Vendor Specific Information: Len=78 <?>
        Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
pcilib: sysfs_read_vpd: read failed: No such device
```
`hciconfig -a`:
```
hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:3 acl:0 sco:0 commands:1 errors:0
        Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1
        Link policy:
        Link mode: PERIPHERAL ACCEPT
```
`lsusb`:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 003: ID 0c45:670b Microdia Integrated_Webcam_HD
Bus 001 Device 004: ID 0a5c:21d7 Broadcom Corp. BCM43142 Bluetooth 4.0
Bus 001 Device 006: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 015: ID 04f3:2012 Elan Microelectronics Corp. Touchscreen
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
`ls /lib/firmware/brcm/ | grep -i BCM43142`:
```
BCM43142A0-0489-e055.hcd
BCM43142A0-0489-e062.hcd
BCM43142A0-0489-e096.hcd
BCM43142A0-04ca-2006.hcd
BCM43142A0-04ca-2007.hcd
BCM43142A0-04ca-2009.hcd
BCM43142A0-04ca-2012.hcd
BCM43142A0-04f2-b49d.hcd
BCM43142A0-04f2-b4a1.hcd
BCM43142A0-0930-021f.hcd
BCM43142A0-0930-0225.hcd
BCM43142A0-0930-0226.hcd
BCM43142A0-0a5c-2167.hcd
BCM43142A0-0a5c-216a.hcd
BCM43142A0-0a5c-216c.hcd
BCM43142A0-0a5c-216d.hcd
BCM43142A0-0a5c-21d3.hcd
BCM43142A0-0a5c-21d6.hcd
BCM43142A0-0a5c-21d7.hcd
BCM43142A0-0a5c-21d8.hcd
BCM43142A0-0a5c-21dc.hcd
BCM43142A0-0a5c-21fe.hcd
BCM43142A0-105b-e065.hcd
BCM43142A0-13d3-3388.hcd
BCM43142A0-13d3-3389.hcd
BCM43142A0-13d3-3427.hcd
BCM43142A0-13d3-3482.hcd
BCM43142A0-13d3-3484.hcd
BCM43142A0-185f-2167.hcd
```
`md5sum /lib/firmware/brcm/BCM43142A0-0a5c-21d7.hcd`:
```
2af7a8c10d9e0cc8dd178d3ad8751a06  /lib/firmware/brcm/BCM43142A0-0a5c-21d7.hcd
```
`lsmod | grep bluetooth`:
```
bluetooth            1114112  15 btrtl,btmtk,btintel,btbcm,bnep,btusb
ecdh_generic           16384  1 bluetooth
rfkill                 40960  6 bluetooth,dell_laptop,dell_rbtn,cfg80211
crc16                  12288  2 bluetooth,ext4
```
`rfkill list`:
```
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

winterheart, 2024-03-27:
Hello.
You have Broadcom WiFi+Bluetooth combined device. Please try to apply these instructions: https://github.com/winterheart/broadcom-bt-firmware?tab=readme-ov-file#notes-about-combined-wifibluetooth-devices

9srp6, 2024-03-27:
@winterheart Thank you for responding
I don't think I  have the nvram file, is it named something different?

`ls /sys/firmware/efi/efivars`:
```
AfuDellMeUpdateReq-01368881-c4ad-4b1d-b631-d57a8ec8db6b
BBSFuncVar-01368881-c4ad-4b1d-b631-d57a8ec8db6b
Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot0004-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot0007-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot0008-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot0009-8be4df61-93ca-11d2-aa0d-00e098032b8c
Boot000B-8be4df61-93ca-11d2-aa0d-00e098032b8c
BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
BugCheckCode-ba57e015-65b3-4c3c-b274-659192f699e3
BugCheckParameter1-ba57e015-65b3-4c3c-b274-659192f699e3
BugCheckProgress-ba57e015-65b3-4c3c-b274-659192f699e3
CapsuleLongModeBuffer-711c703f-c285-4b10-a3b0-36ecbd3c8be2
ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
CurrentPolicy-77fa9abd-0359-4d32-bd60-28f4e78f784b
db-d719b2cb-3d3a-4596-a3bc-dad00e67656f
dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f
DELL_BBSBCV_PTR-0a602c5b-05a0-40c4-9181-edcd891d001c
DELL_BBSIPL_PTR-0a602c5b-05a0-40c4-9181-edcd891d001b
DELL_BBS_PTR-0a602c5b-05a0-40c4-9181-edcd891d0019
DELLGNVS-58a9b489-5978-4775-abb8-e0184f7cbd2f
DELLGNVS_PTR-0a602c5b-05a0-40c4-9181-edcd891d0070
DellHardDriveSize-0a602c5b-05a0-40c4-9181-edcd891d0017
DELL_HDS_PTR-0a602c5b-05a0-40c4-9181-edcd891d0015
DellProduct-0a602c5b-05a0-40c4-9181-edcd891d00f5
DellPwdJumper-0a602c5b-05a0-40c4-9181-edcd891d0040
DellSystemIdExt-01368881-c4ad-4b1d-b631-d57a8ec8db6b
EDID_PTR-0a602c5b-05a0-40c4-9181-edcd891d0060
EfiTime-9d0da369-540b-46f8-85a0-2b5f2c301e15
ErrOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
ErrOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
FirmwareUpdateCountVar-e08126e5-b93c-4f5e-8e81-6dfcf34ff88e
FlashRuleVarFlag-6ab5f341-2e08-4b62-a345-eecff4f03fb2
GNVS_PTR-0a602c5b-05a0-40c4-9181-edcd891d0003
InSetup-0a602c5b-05a0-40c4-9181-edcd891d0035
IsRuntime-01368881-c4ad-4b1d-b631-d57a8ec8db6b
KEK-8be4df61-93ca-11d2-aa0d-00e098032b8c
LoaderDevicePartUUID-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
MonotonicCounter-01368881-c4ad-4b1d-b631-d57a8ec8db6b
OA3DellMSDMvariable-fdad31fb-ef81-49ec-aed2-1ddacca1268d
OemBiosVersion-717576f5-543a-4312-9146-89faac460514
OfflineUniqueIDRandomSeedCRC-eaec226f-c9a3-477a-a826-ddc716cdc0e3
OfflineUniqueIDRandomSeed-eaec226f-c9a3-477a-a826-ddc716cdc0e3
OptVar-ceab3323-daab-92ee-c112-abee5a6ebd2a
OptVar-ceab3323-daab-92ee-c112-abee5a6ebd2b
OsIndications-8be4df61-93ca-11d2-aa0d-00e098032b8c
OsIndicationsSupported-8be4df61-93ca-11d2-aa0d-00e098032b8c
PBRDevicePath-a9b5f8d2-cb6d-42c2-bc01-b5ffaae4335e
PCI_COMMON-aca9f304-21e2-4852-9875-7ff4881d67a5
PK-8be4df61-93ca-11d2-aa0d-00e098032b8c
PlatformLang-8be4df61-93ca-11d2-aa0d-00e098032b8c
PlatformLangCodes-8be4df61-93ca-11d2-aa0d-00e098032b8c
PsaReadyToBoot-fddd33bc-41f7-4d53-bf65-136056e8ccc6
S3SS-4bafc2b4-02dc-4104-b236-d6f1b98d9e84
SecureBoot-8be4df61-93ca-11d2-aa0d-00e098032b8c
SecureFlashSetupVar-35c936af-e1e1-441a-bad1-e1544e9d97a6
Setup-80e1202e-2697-4264-9cc9-80762c3e5863
SetUpdateCountVar-81c76078-bfde-4368-9790-570914c01a65
SetupMode-8be4df61-93ca-11d2-aa0d-00e098032b8c
SignatureSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
SMBIOS_START_ADDR-0a602c5b-05a0-40c4-9181-edcd891d0036
SpdAddrOem-efb14200-e015-11de-8a39-0800200c9a66
Timeout-8be4df61-93ca-11d2-aa0d-00e098032b8c
UnlockIDCopy-eaec226f-c9a3-477a-a826-ddc716cdc0e3
UsbHandOnOffEntry-0a602c5b-05a0-40c4-9181-edcd891d0018
USB_POINT-01368881-c4ad-4b1d-b631-d57a8ec8db6b
V1P0-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P10-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P11-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P12-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P13-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P14-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P15-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P16-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P17-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P18-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P19-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1A-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1B-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1C-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1D-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1E-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P1F-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P20-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P21-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P22-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P23-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P24-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P25-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P26-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P27-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P28-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P29-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2A-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2B-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2C-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2D-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2E-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P2F-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P3-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P4-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P5-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P6-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P7-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P8-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1P9-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PA-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PB-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PC-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PD-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PE-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
V1PF-01bd3876-1ad6-4e59-b39a-7a0b1bde20ac
WriteOnceStatus-4b3082a3-80c6-4d7e-9cd0-583917265df1

winterheart, 2024-03-27:
Can you please post full `dmesg` output? In any case, did you poweroff/poweron (not just reboot) notebook after bluetooth firmware install?

9srp6, 2024-03-27:
>did you poweroff/poweron
>
Did that, nothing changed.
It might be of interest to mention that I tried installing the broadcom-wl-dkms. Following which the bt device disappeared from both the `lsusb` and `rfkill` list. Instead only wlan0 showed up in the in `rfkill`. Had to do a cold reboot to restore everything back to the way it was. Anyways  here is the full `dmesg` output:

https://gist.github.com/9srp6/e34a261749008a6ae13efefacdafdb00

winterheart, 2024-04-05:
Well, that strange.
Guess that device is pretty old and Broadcom didn't test well latest driver for compatibility issues. Can you please also test `BCM43142A0-0a5c-21d7.hcd` from https://github.com/winterheart/broadcom-bt-firmware/releases/tag/v12.0.1.1011_p1 version?

9srp6, 2024-04-05:
> Can you please also test BCM43142A0-0a5c-21d7.hcd from https://github.com/winterheart/broadcom-bt-firmware/releases/tag/v12.0.1.1011_p1 version?

`dmesg | grep -i bluetooth`
```
[   13.268565] Bluetooth: Core ver 2.22
[   13.268590] NET: Registered PF_BLUETOOTH protocol family
[   13.268591] Bluetooth: HCI device and connection manager initialized
[   13.268596] Bluetooth: HCI socket layer initialized
[   13.268598] Bluetooth: L2CAP socket layer initialized
[   13.268601] Bluetooth: SCO socket layer initialized
[   13.613587] Bluetooth: hci0: BCM: chip id 70
[   13.614586] Bluetooth: hci0: BCM: features 0x06
[   13.630599] Bluetooth: hci0: BCM43142A
[   13.630606] Bluetooth: hci0: BCM43142A0 (001.001.011) build 0000
[   13.824452] Bluetooth: hci0: BCM43142A0 'brcm/BCM43142A0-0a5c-21d7.hcd' Patch
[   15.955102] Bluetooth: hci0: command 0xfc4c tx timeout
[   15.955144] Bluetooth: hci0: BCM: Patch command fc4c failed (-110)
[   15.955164] Bluetooth: hci0: BCM: Patch failed (-110)
[   18.088399] Bluetooth: hci0: command 0xfc4c tx timeout
[   18.088429] Bluetooth: hci0: BCM: Reset failed (-110)
```
`hciconfig -a`:
```
hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:538 acl:0 sco:0 events:44 errors:0
        TX bytes:6290 acl:0 sco:0 commands:46 errors:0
        Features: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1
        Link policy:
        Link mode: PERIPHERAL ACCEPT
```
`md5sum /lib/firmware/brcm/BCM43142A0-0a5c-21d7.hcd`:
```
2af7a8c10d9e0cc8dd178d3ad8751a06  /lib/firmware/brcm/BCM43142A0-0a5c-21d7.hcd
```