winterheart/broadcom-bt-firmware #41: Add support for "Broadcom Corp. BCM2070 Bluetooth 2.1 + EDR"

breversa, 2024-04-25:
**Describe the bug**
I believe this device (0a5c:21b4) is not fully supported, as it's unable to see some Bluetooth devices in pairing mode (notable some cheapo iHome Bluetooth speaker).

**Provide info about your system**
```
Linux stephanie-HP-Pavilion-dv6-Notebook-PC 6.5.0-28-generic #29~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr  4 14:39:20 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

[    1.525228] usb 2-1.5: Product: Broadcom 2070 Bluetooth
[   14.351804] Bluetooth: Core ver 2.22
[   14.351836] NET: Registered PF_BLUETOOTH protocol family
[   14.351840] Bluetooth: HCI device and connection manager initialized
[   14.351848] Bluetooth: HCI socket layer initialized
[   14.351852] Bluetooth: L2CAP socket layer initialized
[   14.351861] Bluetooth: SCO socket layer initialized
[   16.452919] Bluetooth: hci0: command 0x1009 tx timeout
[   16.456915] Bluetooth: hci0: Opcode 0x1009 failed: -110
[   31.764733] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   31.764743] Bluetooth: BNEP filters: protocol multicast
[   31.764751] Bluetooth: BNEP socket layer initialized
[ 1087.428353] Bluetooth: hci0: unexpected event for opcode 0x1009
[ 1087.482603] Bluetooth: MGMT ver 1.22
[ 1087.502877] Bluetooth: RFCOMM TTY layer initialized
[ 1087.502891] Bluetooth: RFCOMM socket layer initialized
[ 1087.502901] Bluetooth: RFCOMM ver 1.11
[ 1239.082477] Bluetooth: hci0: Opcode 0x0402 failed: -22

0d:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4313 802.11bgn Wireless Network Adapter [14e4:4727] (rev 01)
	DeviceName: Matador 802.11bgn/BT combo Adapter
	Subsystem: Hewlett-Packard Company BCM4313 802.11bgn Wireless Network Adapter [103c:1483]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at c4400000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: <access denied>
	Kernel driver in use: wl
	Kernel modules: bcma, wl

13:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5209 PCI Express Card Reader [10ec:5209] (rev 01)
hci0:	Type: Primary  Bus: USB
	BD Address: CC:52:AF:A8:A8:4A  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:1267 acl:0 sco:0 events:61 errors:0
	TX bytes:4821 acl:0 sco:0 commands:55 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0x9b 0xff 0x79 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'stephanie-HP-Pavilion-dv6-Notebook-PC'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version: 3.0 (0x5)  Revision: 0x1e1
	LMP Version: 3.0 (0x5)  Subversion: 0x4203
	Manufacturer: Broadcom Corporation (15)

Bus 002 Device 004: ID 0a5c:21b4 Broadcom Corp. BCM2070 Bluetooth 2.1 + EDR
Bus 002 Device 005: ID 0000:0538   USB OPTICAL MOUSE
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 5986:02ac Acer, Inc HP TrueVision HD Webcam
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Please note that I don't have that device readily at hand, so I'll need some time to provide extra information.

winterheart, 2024-04-30:
Hello.
It seems that your device belongs to another family of Broadcom BT devices (2.1) while firmwares in this repository intended to use with 4.0-era devices. I can't find any witnesses that device even works with Linux. Sorry, I can't help there.

closed 2024-04-30 by winterheart

breversa, 2024-04-30:
Damn, that sucks! :-(

Is there anything *I* could do to help?
Outside confirming that this device work at least partially already, as it's able to see *some* Bluetooth devices, but not all.