winterheart/broadcom-bt-firmware #42: Firmware version not showing on dmesg. HP ProBook 6470b. Ubuntu 24.04. BCM20702A0.

danbates2, 2024-05-09:
I'm looking to update the firmware with my onboard my BT device as sound quality is so poor, I've tried many other avenues.. The introductory README.md states a dmesg command used to find current firmware version. My system's not providing the expected output.

```
$ sudo dmesg | grep -i bluetooth

[    6.862113] Bluetooth: Core ver 2.22
[    6.862157] NET: Registered PF_BLUETOOTH protocol family
[    6.862160] Bluetooth: HCI device and connection manager initialized
[    6.862165] Bluetooth: HCI socket layer initialized
[    6.862168] Bluetooth: L2CAP socket layer initialized
[    6.862177] Bluetooth: SCO socket layer initialized
[    8.010732] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.010739] Bluetooth: BNEP filters: protocol multicast
[    8.010745] Bluetooth: BNEP socket layer initialized
[    8.015707] Bluetooth: MGMT ver 1.22
[   10.895177] Bluetooth: RFCOMM TTY layer initialized
[   10.895189] Bluetooth: RFCOMM socket layer initialized
[   10.895199] Bluetooth: RFCOMM ver 1.11
```

My laptop is old.. BIOS settings are for UEFI Native mode and I'm running a fresh install of Ubuntu.

Of interest perhaps is a screenshot of the Firmware Updater app that's part of Ubuntu. This shows the Broadcom BT device and states it's updatable.

[image]

winterheart, 2024-05-13:
Hello. Please post output of these commands:

```
uname -a
lspci -nnvv | grep -A12 Broadcom
hciconfig -a
lsusb
```

danbates2, 2024-05-14:
```
$ uname -a
Linux dfour-HP-ProBook-6470b 6.8.0-31-generic #31-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 00:40:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

$ sudo lspci -nnvv | grep -A12 Broadcom
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43228 802.11a/b/g/n [14e4:4359]
	DeviceName: WLAN
	Subsystem: Hewlett-Packard Company BCM943228HM4L 802.11a/b/g/n 2x2 Wi-Fi Adapter [103c:182c]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 19
	Region 0: Memory at d0400000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
	Capabilities: [58] Vendor Specific Information: Len=78 <?>
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+

$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: F4:B7:E2:FE:8D:32  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:695 acl:0 sco:0 events:50 errors:0
	TX bytes:3174 acl:0 sco:0 commands:50 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT

$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 003: ID 0a5c:21e1 Broadcom Corp. HP Portable SoftSailing
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 003: ID 04f2:b230 Chicony Electronics Co., Ltd Integrated HP HD Webcam
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

note: I had to sudo the lspci command to show the 'capabilities'.

winterheart, 2024-05-14:
OK, you got this device:
```
Bus 001 Device 003: ID 0a5c:21e1 Broadcom Corp. HP Portable SoftSailing
```
According to DEVICES.md, you need `BCM20702A1-0a5c-21e1.hcd` firmware in /lib/firmware/brcm. But I don't see system's attempts to search and load this firmware. That means either Linux kernel don't have support for this device or `btbcm` module is not loaded on system boot. Try these commands:

```
sudo modprobe btbcm
# Diagnose again dmesg output
sudo dmesg | grep -i bluetooth
```

danbates2, 2024-05-14:
The BCM20702A0 chipset I see are BT4.0 only. I don't think a firmware upgrade will provide the BT5.0 support will it? I'm going to bite the bullet and get a BT5.0 or greater USB dongle for my old laptops and new headsets..

I can continue to provide info if you need from my particular setup, I'm curious to find what a firmware upgrade will do if possible and easy enough for all involved..

Cheers.

danbates2, 2024-05-14:
```
$ sudo modprobe btbcm
$ sudo dmesg | grep -i bluetooth
[    6.627182] Bluetooth: Core ver 2.22
[    6.627211] NET: Registered PF_BLUETOOTH protocol family
[    6.627214] Bluetooth: HCI device and connection manager initialized
[    6.627218] Bluetooth: HCI socket layer initialized
[    6.627221] Bluetooth: L2CAP socket layer initialized
[    6.627227] Bluetooth: SCO socket layer initialized
[    7.796829] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.796836] Bluetooth: BNEP filters: protocol multicast
[    7.796842] Bluetooth: BNEP socket layer initialized
[    7.805055] Bluetooth: MGMT ver 1.22
[   10.496018] Bluetooth: RFCOMM TTY layer initialized
[   10.496031] Bluetooth: RFCOMM socket layer initialized
[   10.496039] Bluetooth: RFCOMM ver 1.11
[ 1390.210697] Bluetooth: MGMT ver 1.22
[ 9095.584678] Bluetooth: MGMT ver 1.22
[ 9258.314790] Bluetooth: MGMT ver 1.22
[ 9704.590300] Bluetooth: MGMT ver 1.22
[10025.855016] Bluetooth: MGMT ver 1.22
[17541.937558] Bluetooth: MGMT ver 1.22
```

Perhaps this machine is too old for this kernel?

I'll try load the file directly to the firmware folder.

danbates2, 2024-05-14:
```
@usr-HP-ProBook-6470b:/lib/firmware/brcm$ ls -la | grep BCM20702A1-0a5c-21e1.hcd
-rw-r--r--  1 root root  35763 Oct 10  2022 BCM20702A1-0a5c-21e1.hcd
```

winterheart, 2024-05-14:
> The BCM20702A0 chipset I see are BT4.0 only. I don't think a firmware upgrade will provide the BT5.0 support will it? I'm going to bite the bullet and get a BT5.0 or greater USB dongle for my old [...]

No, firmware will not provide 5.0 capabilities, chipset designed with BT 4.0 spec. Firmware will provide you optimized working with some Bluetooth profiles (clear audio in headphones for example).

After digging Linux kernel code I found, that kernel is not configured to call additional functions for loading external firmwares. This can be relatively easy fixed, but you'll need compile own kernel or kernel module btbcm. If you ready for this, I'll prepare a patch for changes.

winterheart, 2024-05-14:
Here actual patch:

```diff
diff --git a/drivers/bluetooth/btusb.c b/drivers/bluetooth/btusb.c
index e3946f7b736e..d69c82beaece 100644
--- a/drivers/bluetooth/btusb.c
+++ b/drivers/bluetooth/btusb.c
@@ -84,7 +84,7 @@ static const struct usb_device_id btusb_table[] = {
        { USB_DEVICE(0x0e8d, 0x763f) },

        /* Broadcom SoftSailing reporting vendor specific */
-       { USB_DEVICE(0x0a5c, 0x21e1) },
+       { USB_DEVICE(0x0a5c, 0x21e1), .driver_info = BTUSB_BCM_PATCHRAM },

        /* Apple MacBookPro 7,1 */
        { USB_DEVICE(0x05ac, 0x8213) },
```

danbates2, 2024-05-14:
Good find.
I'm not familiar with patching kernels or modules. I can compile, sure.
If this is a one off patch for the firmware update then I'm up for it, but if it needs repeating each kernel update, then I'll leave it be.

winterheart, 2024-05-14:
I'm planning to upstream contribution to kernel code base, but I need some confirmations that these changes actually works as intended. I don't have such hardware on my reach, so I cannot test it myself.

danbates2, 2024-05-14:
Can you provide a rough layout to this test? What's involved?

winterheart, 2024-05-14:
You need build and load own kernel with applied patch (rough guide for Ubuntu: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel).

After booting on modified kernel you should receive dmesg output about firmware actually loaded (your output will be different):
```
[    6.596460] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
[    6.600992] Bluetooth: hci0: BCM20702A1 'brcm/BCM20702A1-0b05-17cb.hcd' Patch
[    7.343460] Bluetooth: hci0: BCM: features 0x07
[    7.359539] Bluetooth: hci0: Broadcom Bluetooth Device
[    7.359542] Bluetooth: hci0: BCM20702A1 (001.002.014) build 1467
```

After that I expects that you'll get additional sound high-quality A2DP profiles related to bluetooth and overall enhancing working of Bluetooth stack (like stable discovery and pairing).